kiibohd/controller #224: 'None' in kll file cause error when build

botaski, 2017-10-18:
Putting 'None' in kll (for disabling specific key) ALWAYS cause error when buiding custom firmware.
Example as follows,
 
in kll/layouts/whitefox/whitefox.kll
`U"Right"      :   None;`

Here's the output...

botaski@ubuntu:~/git/controller/Keyboards$ ./whitefox.bash 
linux-gnu/Linux is supported.
Selected Generator: Unix Makefiles
linux-gnu.WhiteFox.gcc.make
-- Build CPU Detected:
x86_64
-- Build OS Detected:
Linux 4.4.0-93-generic
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c;Lib/entropy.c;Lib/time.c
-- Bootloader Type:
dfu
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc 
-- Detected Scan Module Source Files:
Scan/Devices/ISSILed/i2c.c;Scan/Devices/ISSILed/led_scan.c;Scan/Devices/MatrixARM/matrix_scan.c;Scan/WhiteFox/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c;Macro/PixelMap/pixel.c;/home/botaski/git/controller/Keyboards/linux-gnu.WhiteFox.gcc.make/generatedPixelmap.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Detected Layout Files:
/home/botaski/git/controller/Scan/Devices/ISSILed/capabilities.kll
/home/botaski/git/controller/Scan/Devices/MatrixARM/capabilities.kll
/home/botaski/git/controller/Macro/PartialMap/capabilities.kll
/home/botaski/git/controller/Macro/PixelMap/capabilities.kll
/home/botaski/git/controller/Output/HID-IO/capabilities.kll
/home/botaski/git/controller/Output/pjrcUSB/capabilities.kll
/home/botaski/git/controller/Scan/WhiteFox/scancode_map.kll
/home/botaski/git/controller/Scan/WhiteFox/scancode_map.truefox.kll
/home/botaski/git/controller/kll/layouts/stdFuncMap.kll
/home/botaski/git/controller/kll/layouts/whitefox/whitefox.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /home/botaski/git/controller/Keyboards/linux-gnu.WhiteFox.gcc.make
[  3%] Generating KLL Layout
kll BETA 0.5c.f0f1032e4690f103be99d616c17d0c22a2a7633c - 2017-10-14 15:44:22 -0700
Traceback (most recent call last):
  File "../../kll/kll", line 159, in <module>
    control.process()
  File "/home/botaski/git/controller/kll/common/stage.py", line 123, in process
    stage.process()
  File "/home/botaski/git/controller/kll/common/stage.py", line 2827, in process
    self.emitter.process()
  File "/home/botaski/git/controller/kll/emitters/kiibohd/kiibohd.py", line 671, in process
    self.fill_dict['ResultMacros'] += "{0}, ".format(self.result_combo_conversion(combo))
  File "/home/botaski/git/controller/kll/emitters/kiibohd/kiibohd.py", line 249, in result_combo_conversion
    cap_index = self.capabilities_index[identifier.name]
KeyError: 'None'
CMakeFiles/kiibohd.dir/build.make:70: recipe for target 'generatedKeymap.h' failed
make[2]: *** [generatedKeymap.h] Error 1
CMakeFiles/Makefile2:264: recipe for target 'CMakeFiles/kiibohd.dir/all' failed
make[1]: *** [CMakeFiles/kiibohd.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Error in make. Exiting...

Enviroment: Ubuntu and macOS
this also happen infinity ergodox, not just whitefox.

Any information would be helpful.
Thanks in advance.

haata, 2017-10-30:
I've reproduced this. Working on a fix.

haata, 2017-10-31:
Should be fixed with: https://github.com/kiibohd/kll/commit/aad5145b12c0cb8dd207d4c00585a9c71256612d

closed 2017-10-31 by haata