kiibohd/controller #225: Caps-, Num-, ScrollLock indicator LED's?

abrahamssonmathias, 2017-10-20:
Hi! I've read through the documentation and the source code to the best of my abilities, but I can't seem to find where to define the I/O for the lock modifier indicator LED's. Could anyone please point me in the right direction?

abrahamssonmathias, 2017-10-25:
Should add that I have so far been using the Teensy 3.2 guide on defining the matrix and mapping keys.

cookiesowns, 2017-11-01:
Also interested in this. As far as I'm currently aware there's no way to have a visual indication that these modifiers are activated.

Kody-Quintana, 2019-02-02:
Is this still not possible?

haata, 2019-02-02:
Not quite yet, the next firmware release should include support for indicator LEDs (I've been fixing quite a few bugs related to lock LED lissues).
https://github.com/kiibohd/controller/issues/313

I'll also be tracking this issue here for Kira: https://github.com/kiibohd/controller/issues/314

haata, 2019-02-02:
Yes! That code already works for turning the LEDs on (just not turning the LEDs off, haha).