kiibohd/controller #234: K-Type Menu button not working

skuko, 2017-10-31:
as the subject says, menu button is not functional. the LED is lit, but the button does not actually do anything. 

verified but multiple people (in massdrop discussion). tested with different switches to eliminate the possibility of a defect switch.

also persists after flashing the "no animation" firmware.

JasonIRL, 2017-11-02:
@skuko I was able to resolve the issue by changing it from "menu" to "app" in the configurator. Does that work for you?

skuko, 2017-11-02:
Yeah, that worked for me too, cheers

haata, 2017-11-15:
The default K-Type layout now uses "app".

https://github.com/kiibohd/controller/commit/630594a537c5d740557fb8ded93bfcf901ee249f

Configurator has also made this switch.

closed 2017-11-15 by haata