kiibohd/controller #237: Dropped keys in 6KRO mode

jalaziz, 2017-11-03:
I'm using macOS 10.13.1.

When using the K-Type in 6KRO mode (via both USB or Bluetooth adapter), I find that keys are dropped randomly when typing. The drops are intermittent, but when they do occur, hitting the same key multiple times usually won't register for a second or so.

The K-Type works perfectly fine in NKRO mode though.

haata, 2017-11-03:
Implement this: https://geekhack.org/index.php?topic=14290.msg307392#msg307392

Kiyiko, 2017-11-08:
Adding to this, though I'm not 100% sure if my observation is the same bug as yours. My observation is non-random.

If I press three keys in rapid succession, the 2nd key remains stuck on until pressed a second time.

I also have no issues in nkro, though I haven't experienced any problems with pressing a single key multiple times.

windows 10, k-type

[image]

haata, 2017-11-08:
I'm working on multiple fixes right now to help with this. It's my 1st priority atm.

sirideain, 2017-11-15:
This sounds similar to an issue I am seeing where keys will sometimes not register. Unplugging the keyboard and plugging it back in fixes the issue. Rebooting will bring it back. I am only seeing it on my laptop though and not my desktop. Both are running Windows 10.

I used the Switch Hitter utility (http://elitekeyboards.com/switchhitter.php) to see what the keys were doing. Something I noticed is that sometimes a key would register for a DOWN press, but then would appear stuck DOWN in software. It would not be physically stuck down. Hitting the key again would register as an UP. And while it was stuck down, it wasn't constantly being pressed either. That would explain why sometimes key strokes were not registering. You can see an example below for the letter W.

55:16.0650 Q (0x51, BIOS 0x10) DOWN
55:16.0713 W (0x57, BIOS 0x11) DOWN
55:16.0783 E (0x45, BIOS 0x12) DOWN
55:16.0807 Q (0x51, BIOS 0x10) UP -> 157ms
55:16.0813 E (0x45, BIOS 0x12) UP -> 29ms
55:16.0931 R (0x52, BIOS 0x13) DOWN
55:17.0114 R (0x52, BIOS 0x13) UP -> 182ms
55:20.0676 W (0x57, BIOS 0x11) UP -> 3964ms

XrXr, 2017-11-19:
Entering flash mode then exiting (pressing the combination twice basically) fixes it for me also. A bit nicer than unplugging.

haata, 2017-11-20:
I believe I've fixed nearly all dropped key issues with this commit https://github.com/kiibohd/controller/commit/a9e8619037bde4d27ca11303605a4957054bd872

Still working on the HID IDLE timeouts for some cases on macOS.

XrXr, 2017-11-20:
I just tried flashing my k-type using a build made with 5562633eb26839c5159efbc60f829c37464231ff without any modification. Some keys don't work (backspace, delete) and after typing letters for a few seconds the rainbow animation freezes and no keys work. Not sure if I did anything wrong.
```
[Keyboards]# ./k-type.bash
...
alan ~/k/f/c/Keyboards (master)> sudo dfu-util -D linux-gnu.K-Type.gcc.make/kiibohd.secure.dfu.bin 
[sudo] password for alan: 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1c11
Match product ID from file: b007
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        63288 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
```

haata, 2017-11-21:
kk, I've had some other reports as well on this. Will try to look for a fix this evening.

haata, 2017-11-22:
Ok, the rest of the fixes should be in.

IDLE timeout fix for macOS https://github.com/kiibohd/controller/commit/cb040718ae8af8272e2bb30791f4186f4a25c74f

closed 2017-11-22 by haata

bam80, 2019-01-28:
> When using the K-Type in 6KRO mode (via both USB or Bluetooth adapter)

AFAIK, K-Type doesn't have Bluetooth, so what Bluetooth adapter is meant? Is it just USB to Bluetooth adapter?