kiibohd/controller #240: Modifier keys are broken in simple macro

botaski, 2017-11-07:
Hi,
Chalacters placed after uppercase letters and symbols(there are normally input by combination of shift key) make following letters all uppercase in kll file.

Example is as follows...(i tried kll/layout/whitefox/whitefox.kll)
**U"LEFT" : 'Hello, World';
U"RIGHT" : 'foo@bar.com';**

then the result is ...
**LEFT key --> HELLO<WORLD
RIGHT key --> foo@BAR>COM**

any information will be appreciated.

Kiyiko, 2017-11-09:
My experience with this is not the same. On my k-type, your kll returns **HELO< WRD** and **fo@BAR>CM**

My understanding of the issue is that these commands do not type out the letters one-by-one, but rather press all the keys at more-or-less the same time, resulting in the shift key affecting all subsequent keystrokes. This also results in characters being able to be used only once, which seems to be the main difference between what I see and what you reported.

InfinityTotality, 2018-02-13:
It doesn't seem to be a timing issue, but rather the keyboard does not send any key up events until the end of a sequence. The result is that each key can only be pressed once and will be held until the sequence is completed.