kiibohd/controller #242: Keyboard (KType) occasionally becomes unresponsive (and animations stop)

intafon, 2017-11-15:
Occasionally, the keyboard animations will stop, at which point the entire keyboard becomes unresponsive for a second or two. When the keyboard comes back online, the animation resumes at its initial state.

Yesterday, it seemed to be triggered by double-clicking the tab key while connected to an ssh session.

Connected to: (macbook pro, retina 15 inch, late 2013, keyboard connected via thunderbolt monitor usb port)

Connected via: issue occurs connected to both plugable usb2.0 10port hub (connected to thunderbolt monitor hub) and when connected directly to thunderbolt monitor usb hub.

Trying out connecting the keyboard directly to the macbook today to see if the issue recurs.

tengg, 2017-12-04:
The volumes keys will crash the keyboard 80% of the time too..

smasher816, 2018-04-04:
Unfortunately the ISSI chips can be a tad unstable. Newer revisions of the firmware may help aleviate this issue.

https://github.com/kiibohd/controller/commit/5790ebc41f160b872c06e596acdb28807063e91f
"Added auto-reset logic to I2C bus whenever it locks up"