kiibohd/controller #247: K-type build broken on Linux

XrXr, 2017-11-27:
Building off of 9a0ac1b7a9cbe252a0590f13b4ed3c5362383cda on Arch Linux
```
linux-gnu/Linux is supported.
Unix Build
Selected Generator: Unix Makefiles
linux-gnu.K-Type.gcc.make
-- Build CPU Detected:
x86_64
-- Build OS Detected:
Linux 4.13.12-1-ARCH
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk2x
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/kinetis.c;Lib/delay.c;Lib/entropy.c;Lib/periodic.c;Lib/time.c
-- Bootloader Type:
dfu
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc 7.2.0
-- Detected Scan Module Source Files:
Scan/Devices/ISSILed/i2c.c;Scan/Devices/ISSILed/led_scan.c;Scan/Devices/MatrixARMPeriodic/matrix_scan.c;Scan/Devices/PortSwap/port_scan.c;Scan/Devices/UARTConnect/connect_scan.c;Scan/K-Type/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c;Macro/PixelMap/pixel.c;/home/alan/keyboard/flashing/controller/Keyboards/linux-gnu.K-Type.gcc.make/generatedPixelmap.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/latency/latency.c;Debug/led/led.c;Debug/print/print.c
-- Could NOT find Ctags (missing: CTAGS_EXECUTABLE) 
-- Detected Layout Files:
/home/alan/keyboard/flashing/controller/Scan/Devices/ISSILed/capabilities.kll
/home/alan/keyboard/flashing/controller/Scan/Devices/MatrixARMPeriodic/capabilities.kll
/home/alan/keyboard/flashing/controller/Scan/Devices/PortSwap/capabilities.kll
/home/alan/keyboard/flashing/controller/Scan/Devices/UARTConnect/capabilities.kll
/home/alan/keyboard/flashing/controller/Macro/PartialMap/capabilities.kll
/home/alan/keyboard/flashing/controller/Macro/PixelMap/capabilities.kll
/home/alan/keyboard/flashing/controller/Output/HID-IO/capabilities.kll
/home/alan/keyboard/flashing/controller/Output/pjrcUSB/capabilities.kll
/home/alan/keyboard/flashing/controller/Debug/latency/capabilities.kll
/home/alan/keyboard/flashing/controller/Scan/K-Type/scancode_map.kll
/home/alan/keyboard/flashing/controller/kll/layouts/k-type/release.1.kll
/home/alan/keyboard/flashing/controller/kll/layouts/stdFuncMap.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alan/keyboard/flashing/controller/Keyboards/linux-gnu.K-Type.gcc.make
[  2%] Generating KLL Layout
kll BETA 0.5c.aad5145b12c0cb8dd207d4c00585a9c71256612d - 2017-10-31 00:26:12 -0700
Usage: kll [options..] [<generic>..]
kll: error: unrecognized arguments: --kiibohd-debug
make[2]: *** [CMakeFiles/kiibohd.dir/build.make:74: generatedKeymap.h] Error 2
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/kiibohd.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Error in build. Exiting...
```
```

haata, 2017-11-27:
You'll need to update the kll git repo.

XrXr, 2017-11-27:
It works now! thanks :D

closed 2017-11-27 by XrXr