kiibohd/controller #248: Gmail ignoring Shift key when trying to select multiple emails

natemcintyre, 2017-11-27:
After flashing my firmware today (ironically just to change LED colors) I noticed that I couldn't bulk select emails via the "Shift + Click" action (needed to get rid of lots of Cyber Monday spam).

Can confirm that the issue was resolved when hid idle is set to 0 rather than the default 24ms (see Discord conversation thread below).

Discord Conversation:
```
[4:07 PM] d3vnull: Ok, when connected to that with outputDebug it was basically spamming
[4:07 PM] d3vnull: I would assume that is our 24ms send
[4:07 PM] HaaTa: yep
[4:07 PM] HaaTa: that's what hid idle does
[4:08 PM] HaaTa: I may add some code to make that less spammy with hid idle enabled
[4:08 PM] d3vnull: idle:
INFO - USB Idle Config: 24 ms - 6
[4:08 PM] HaaTa: yep, that's the default
[4:08 PM] HaaTa: for macOS
[4:08 PM] d3vnull: k
[4:08 PM] HaaTa: just set to 0 to disable entirely
[4:09 PM] d3vnull: yup, just did that
[4:09 PM] d3vnull: that fixed it
```

haata, 2017-11-27:
This is macOS High Sierra, NKRO mode.

Likely a caching issue with HID IDLE enabled.

sebkouba, 2017-11-28:
I have the same issue now on Sierra but managed the workaround...

RotsiserMho, 2017-11-28:
I too had this problem over the weekend. Is a fix forthcoming? I had to revert to an older firmware.

haata, 2017-11-28:
Should be fixed with https://github.com/kiibohd/controller/commit/86aecdd572186fed6ca5fb583af746ecca6b9ce6

Please let me know if there are other issues.

closed 2017-11-28 by haata