kiibohd/controller #250: Ergodox-Infinity keys stuck/repeating often with latest build

dfaught, 2017-12-04:
Keys are often repeating.  It appears as though most I can interrupt them but sometimes I have to unplug the keyboard to stop the repeat.  I pulled this morning after the latest changes.

Arch Linux / ED-Infinity.

haata, 2017-12-04:
Are the repeating keys coming from both sides?

Did you reflash both sides of the ergodox? (I've been doing lots of
debounce changes lately).

dfaught, 2017-12-04:
I did reflash both sides.  I have seen the behavior on both sides but feel it is more common on the slave.  I noticed it first a couple of days ago when I pulled and flashed, but it was quite intermittent.  I pulled and flashed both sides this morning and it occurred to frequently for me to use it.  I've rolled back to firmware produced by the configurator.

If there's debug info I can get you let me know what you need.  I should have time this afternoon to play with it.

haata, 2017-12-04:
I've reproduced it, seems to be a new bug with 6kro mode.

haata, 2017-12-04:
Quick question, did you try to switch between 6kro and nkro modes manually?
(I think I know what's going on, just trying to get more info)

dfaught, 2017-12-04:
I haven't tried switching, in fact I don't have 6/N-KRO mapped.  I can add it and test in each state if you feel it would be helpful

haata, 2017-12-07:
This should be fixed now.

https://github.com/kiibohd/controller/commit/3b89b8a26efe47fe2a7a807741f726f79b235a6e

closed 2017-12-07 by haata

dfaught, 2017-12-09:
Still happens.  Now with an extra bonus of slave side lockup until power cycled.

AMcBain, 2019-06-25:
I have a WhiteFox (Kickstarter) keyboard and occasionally keys seem to get stuck and repeat. This didn't seems to happen much more after I updated to the Configurator v1 and reflashed my keyboard to try a layout tweak. Unfortunately as the cause is random (at least to me, for now) I don't have any steps to replicate the issue.

Unlike dfaught I can cancel the repetition by hitting the repeating key a second time. The key that repeats varies so it's not an issue (as far as I'm aware and can see) with the key switches.

I don't know if the Configurator v1 shipped with a new Controller version/compiler but the symptoms match this ticket. If you'd rather I open it against the Configurator I can go do so there.

I noticed this issue before Configurator v1 but it was limited to some accidental key combination I can't purposely replicate resulting in triggering the "volume down" (normally fn+page dn; layer 1) mapping repeatedly, sending my system volume down to zero. In that case, pressing "volume up" (fn+page up; layer 1) would cancel the effect. I narrowed it to repetition because trying to raise the volume again through the Windows tray icon popup is rapidly (but not instantly) reduced to zero again.

I love this keyboard and have some tech savvyness so if you have things I can try to help narrow this down if they're not super complicated I'm willing to try them.

cerealcable, 2024-03-15:
A friend of mine were talking and this exact issue came up. We both have been experiencing it on our whitefox and nightfox boards. I've been more or less just ignoring it since it's only occasionally being a problem but just commenting so others know its still happening.