kiibohd/controller #252: k-type plugs in to Mac, creates a network interface

tengg, 2017-12-09:
After a few days of use, there's 200+ network service interface in System Preference -> Network..

Any idea why?

haata, 2017-12-09:
This is bug/feature of macOS where it creates a network device for each
serial interface it sees.
The K-Type has a virtual serial port which can be used for debugging (it's
an interactive cli).

tengg, 2017-12-09:
@haata Is there a workaround on the OS side then?

manno, 2018-02-05:
This should go away once the debug console is disabled by default: https://github.com/kiibohd/controller/issues/159

haata, 2018-02-05:
@tengg I did some looking into this, but I couldn't find anything with my google-fu (I'm not really a macOS expert).

@manno Yeah, I am planning on disabling the debug console. It will be a while though as it will require HID-IO integration to replace some of the functionality.

haata, 2019-02-18:
HID-IO is starting to come together (finally).