kiibohd/controller #253: K-type cannot function with more than one 'fingerprints'/'fingerprints_two_tone' animation

lambricm, 2017-12-10:
I was playing around with the configurator and ended up making my keyboard unusable (no lights after flash mode was complete, no keypresses detected, unplugging/replugging in didn't work, and restarting my computer did nothing).

I messed around with the configuration and found out that if you flash 2 versions of the 'fingerprints' animation, it will have these issues.

akatrevorjay, 2017-12-12:
Hm, I'm surprised you can get a single one to not crash on init. I can't even have one ;)

haata, 2018-02-27:
I believe this was an issue with the Result macro event queuing.
Should be fixed with https://github.com/kiibohd/controller/commit/ac024ca70826040d2a48c688232c52bed13fb828.

Please re-open if it's still broken.

closed 2018-02-27 by haata