kiibohd/controller #255: Fix washed-out LED colors with gamma expansion

indecisivemal, 2017-12-12:
Colors on my new K-Type were looking awfully washed out and in some cases just plain wrong. It seems that no gamma expansion is applied to user-provided values. They look much better after manually expanding them, but this should be done by the software.

Basically: When taking the input value from the configurator it should put the value in the range 0-1, raise it to the power of $gamma, then put it back to 0-255 (or whatever range is used internally)

What I've been doing manually:
- Target color is `RGB 255,143,0` (orange). Using this value results in a yellow keyboard.
- I've been using `gamma = 2.2` which looks good. I haven't bothered to try others.
- `linear_value = round(((color_value/255)**gamma)*255)`
- The resulting values I put into the configurator is `RGB 255,71,0` (basically red on monitor because of double gamma expansion)
- The resulting color on the keyboard is a nice orange, very close to the target color.

The best place to do this is probably when parsing the json, so that color values entered by hand are represented correctly. I apologize if that's not in this repository, I'm not familiar with them.

haata, 2017-12-12:
Neat!

I'm not sure how quickly I could add this to the KLL compiler, but @jbondeson may be able to do something more easily with the configurator.

haata, 2018-12-03:
Now implemented!

https://github.com/kiibohd/controller/commit/a16ee2c3a6ea9dce1c189f14ce8ebd0ea845a659

closed 2018-12-03 by haata

haata, 2018-12-22:
@bradyjoh interesting, I'll take a look. I haven't done much testing off gamma correction on Kinetis yet (mostly on SAM4S/Kira).

haata, 2018-12-22:
Flickering confirmed, gonna try a few things.

haata, 2018-12-23:
@bradyjoh should be fixed now.

bradyjoh, 2018-12-23:
Just flashed, I too can confirm it's not flickering when Gamma is enabled and set to 1.0 or 2.2.

HOWEVER, I might be going crazy but I think that Gamma disabled now causes flickering

haata, 2018-12-23:
Ooops, one sec.

haata, 2018-12-23:
Now try.

bradyjoh, 2018-12-23:
Yep it looks good in both modes now. Thanks @haata ! 

To others that make it here, I just figured out that instead of re-flashing to enable/disable to test Gamma, there's a handy shortcut to use. In a KLL configuration, you can make a map like `U"L" : gamma(2);` where the 2 input triggers Gamma on and off when you tap the L key. That change (and the other inputs to the function) can be seen at the line linked [here](https://github.com/kiibohd/controller/commit/a16ee2c3a6ea9dce1c189f14ce8ebd0ea845a659#diff-314727d6fc232d4dd7634c5d815fb3d6R19)