kiibohd/controller #256: Using a numeric range in pixel animation only lights last LED in range.

AevumDecessus, 2017-12-21:
* **Device**:
K-Type (Massdrop November 2017)

* **Firmware Origin**:
Linux Configurator 0.4

* **OS**:
Ubuntu 16.04 LTS

* **Version**:
bcdDevice            2.cb

* **Reproduction Steps**:* 
  * Create new visualization:
  * name: enter_underlight
  * settings: 
      ```framedelay:50, framestretch, loops:1, replace:basic```
  * code:
    ```kll
    P[88-119](255,0,0);
    ```
  * Add "enter_underlight" animation start to the enter key
  * Flash keyboard

* **Expected output**:
  * Upon pressing the enter key, all underlight LED's should flash red momentarily.

* **Resulting Issue**:
  * Upon pressing the enter key, only LED 119 (the bottom right underlight LED) flashes red momentarily

jeremy0x4a, 2018-02-06:
I think it's actually only using the first in the range, not the last. The same happens when using a list (e.g. 88,89,90...) also, only the first is used.

@AevumDecessus The bottom-right LED is 88, continuing clockwise, with 119 being on the right side just above 88. See: [https://kll.wiki/index.php?title=K-type_LED_map](https://kll.wiki/index.php?title=K-type_LED_map)

AevumDecessus, 2018-02-06:
Good catch. Still a problem, but I've updated the description to be accurate.