kiibohd/controller #257: mapping and using fn+right arrow breaks base layer functionality

keybeejeebee, 2017-12-23:
**Device**:
K-Type (Massdrop November 2017)

**Firmware Origin**:
Desktop Configurator

**OS**:
Windows 10

**Version**:
I don't know what to put here. Using configurator 4.1, recent flash.

**Reproduction Steps**:

1. Change to layer 1
2. Map the right arrow to any function or key 
3. Flash Keyboard
4. Press fn+right arrow

**Resulting Issue**:
Base layer functionality no longer works. Some layer 1 functions continue to work, as I can volume up and down, and enter flash mode.

**notes**:
I tried using the terminal debugger commands, and notice that when you press the fn+right arrow you get a repeating sequence of:
Sw H 0:95
Sw H 0:90
about 100 lines or so (totally guessing), and future key presses of any key result in similar output but with different numbers. This behavior is not seen prior to pressing the fn+right arrow combination. I have no idea if this helps any.

kj800x, 2018-01-27:
I'm able to reproduce this issue on my K-Type. It appears to only affect layer 1, as when I tested a layer 2 + right arrow binding it worked correctly (bind fn to _f2_, bind right arrow on layer 2 to 'b').

I am also using: K-Type (Massdrop November 2017), Desktop Configurator 4.1, Windows 10

haata, 2018-01-30:
I believe this is fixed in git master. Still working on getting this
published to the configurator.

eaigner, 2018-02-15:
Happens to me on macOS. Pressing f1+any arrow completely hangs the keyboard and i have to reflash to get i working again.