kiibohd/controller #262: issue with cliFunc_version

slyrat, 2018-01-03:
* **Device**:
Infinity Ergodox

* **Firmware Origin**:
http://configurator.inputclub.com/?layout=MDErgo1-Default
and slight edit of kll files. Issue is in building said kll files

* **OS**:
Windows 10

* **Version**:
Windows: 10 with cygwin

* **Reproduction Steps**:
I've edited a kll file generated from here:
http://configurator.inputclub.com/?layout=MDErgo1-Default
adding the following:
U"F14" : U"Ctrl" + U"C";
U"F15" : U"Ctrl" + U"X";
U"F16" : U"Ctrl" + U"V";

After running the ergodox.bash it gets to around 54% and then fails. See Resulting issue for the output.

* **Resulting Issue**:
This is what I see:
D:/kiibohd/Debug/cli/cli.c: In function 'cliFunc_version':
D:/kiibohd/Debug/cli/cli.c:714:56: error: expected expression before ')' token
  printInt16( CLI_RevisionNumber - CLI_VersionRevNumber );
                                                        ^
make[2]: *** [CMakeFiles/kiibohd.dir/build.make:790: CMakeFiles/kiibohd.dir/Debug/cli/cli.c.obj] Error 1
make[2]: Leaving directory '/cygdrive/d/kiibohd/Keyboards/cygwin.Dvorak-L.gcc.make'
make[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/kiibohd.dir/all] Error 2
make[1]: Leaving directory '/cygdrive/d/kiibohd/Keyboards/cygwin.Dvorak-L.gcc.make'
make: *** [Makefile:84: all] Error 2

haata, 2018-01-04:
Interesting, this may be due to a recent change.
I'll try to look at this sometime this week.

slyrat, 2018-01-04:
Also, I should note that this happened even if I used an un-edited file from the configurator.

haata, 2018-01-06:
This should be fixed by https://github.com/kiibohd/controller/commit/5f8f25e001e1d4aae104d2fa7c6823cae9b180aa

closed 2018-01-06 by haata