kiibohd/controller #266: Ergodox Lock key issues, unlock from a layer working oddly

grizzlybus, 2018-01-09:
I reliably am finding this problem:

With the ergodox infinity (both Massdrop 2016s), using the input.club latest configurator, the default "lock-1" and "lock-2" keys are acting oddly on two different ergodox infinity keyboards, for Mac and Windows 10:

Press the Lock-1 key locks in layer 1 correctly.
Press the Lock-1 key to unlock does not work, UNLESS pressed 3 times.

Press the Lock-2 key locks in layer 2 correctly
Press the Lock-2 key to unlock fails.  No amount of pressing "Lock-2" unlocks this key.
Requires unplugging master to fix.

This is with the left side master on both keyboards or right side master.

I did not have any such issues with earlier versions of the firmware, this started when I did a flash in December 2017.

grizzlybus, 2018-01-09:
These problems do NOT present themselves in the LTS version FYI

bbaydar, 2018-10-01:
Did you ever find a fix for this? Happens to me too.

grizzlybus, 2018-10-03:
> I'm having the same problem with "LOCK2".

Yeah, my solution was to switch from a cutting edge to a "stable" revision
firmware... all my problems went away.

nsuffield, 2018-10-03:
Sorry to be a bit of a dunce ... I'm just using the downloadable configurator from https://input.club/configurator-setup/. Where did you get the stable firmware?

grizzlybus, 2018-10-03:
Assuming you're using https://input.club/configurator-ergodox/ to create
your configuration, there is an option that by default is "latest", but for
me... "lts" works alot better:

[image: image.png]

nsuffield, 2018-10-03:
Actually no, I was using the downloadable app (Kiibohd Configurator) but I have another look at LTS on the online configuration tool. 
Actually ... I'm also tempted to have a look at QMK.

MarkerDave, 2019-09-24:
I also have the same problem, I'm using the Kiibohd Configurator app v1.0.2.

grizzlybus, 2019-09-24:
I found using a stable version fixed this for me- I’ll check the version
when I can...

haata, 2020-03-24:
Please take a look at the latest configurator firmware. I have fixed a bunch of issues.

grizzlybus, 2020-04-06:
@haata - I appreciate you fixing the firmware!  I'm about 10 minutes in, but v1220 (I think?) was a complete fix!  So far so good, I'll report if I see any issues, but the keyboard is typing atleast as well as my 2018 web configurator firmware

closed 2020-04-06 by grizzlybus