kiibohd/controller #267: cli.c.o Operator && has no right operand

LaikeSF, 2018-01-16:
* **Device**:
Ergodox Infinity

* **Firmware Origin**:
git.repo

* **OS**:
Windows 10 Bash

I have been trying to compile using Windows 10 bash shell. I am getting the following error when I try to compile any firmware, both using the stock layout or my own.

* **Resulting Issue**:
[  2%] Building C object CMakeFiles/kiibohd.dir/Debug/cli/cli.c.o
../../Debug/cli/cli.c: In function 'cliFunc_version':
../../Debug/cli/cli.c:714:47: error: operator '&&' has no right operand
 #if CLI_RevisionNumber && CLI_VersionRevNumber
                                               ^
make[2]: *** [CMakeFiles/kiibohd.dir/Debug/cli/cli.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.dir/all] Error 2
make: *** [all] Error 2
Error in build. Exiting..

haata, 2018-01-16:
Have pulled the latest sources from git? (An issue like this was fixed
recently).

LaikeSF, 2018-01-16:
I've tried this across two different builds over the last several days. I pulled the latest source just before I copied the error this morning.

isabelacmor, 2018-01-17:
I'm also seeing this using the latest version of the repo on MacOS. This is even when running the standard `./whitefox.truefox.bash` without any of my custom klls.

haata, 2018-01-17:
@isabelacmor do you use MacPorts or Brew?

isabelacmor, 2018-01-18:
@haata I'm using Brew.

haata, 2018-01-18:
Cool, that's what I'm using as well. Let me test out a fresh clone.

haata, 2018-01-18:
I've verified the fix on my own macOS box with Brew.

This would happen on github forks without any tags set (git computations get confused).
Let me know if there are any more issues.

closed 2018-01-18 by haata

isabelacmor, 2018-01-19:
Confirming that this fix works when cloning the repo, but not when running in a downloaded and unzipped copy.

Thanks for the fix!