kiibohd/controller #270: flashMode no longer working

patrickdepinguin, 2018-01-18:
* **Device**: IC60_LED   (Alphabet; Massdrop)

* **Firmware Origin**: controller.git repo, base commit 5db4e34a69fcc3863b1186900fecb18574048492.
Previously used commit without issue: 65252776370e4f88c4db1bf1092a14723864d14d

kll commit: kiibohd/kll@afdd8574c820bfe11578ca492dbe3e4f7ea10490
Previously used commit without issue: kiibohd/kll@ba099303d96e8b2e492d17ecc02841d440577303

* **OS**: Gentoo Linux

* **Version**: bcdDevice            3.04

* **Reproduction Steps/issue**:
I have following mapping in my layout:
U"ESC" : flashMode();
which originally came from the web interface. It worked for me before, but after updating the controller and kll repo, it no longer works. It seems to hang for about a second (no keys can be typed) then the keyboard seems reset (back in the base layer).
When running dfu-util to flash during this 'hang' period, output is:

```
dfu-util -D linux-gnu.IC60_LED_custom.gcc.ninja/kiibohd.dfu.bin                                                                                                                       
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1c11
Match product ID from file: b007
dfu-util: No DFU capable USB device available
```

Using the flash button still works, output in that case is:
```
dfu-util -D linux-gnu.IC60_LED_custom.gcc.ninja/kiibohd.dfu.bin                                                                                                                       
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1c11
Match product ID from file: b007
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        55316 bytes
Download done.
dfu-util: unable to read DFU status after completion
```

Is this a feature or a bug?
I read about the Secure firmware method, which is not supported by my bootloader. Could it be related?
Is this caused by `flashModeEnabled` ? Has its default value changed? 
I tried setting it to 1 in my kll files but it makes no difference.

haata, 2018-01-18:
After setting `flashModeEnable = 1;`, have you tried using the cli command `reload`?

I'll need to look into this further as I haven't seen any recent reports of this.

patrickdepinguin, 2018-01-19:
What I did was add 'flassModeEnabled = 1;' inside each of my layer kll files, then rebuild the firmware and flashed it. I don't fully understand where in this process a reload should be tried. Do you mean I should connect to the CLI after the flashing, then execute reload?
How could this change things? Is there some persistent state which is preserved even after flashing?

Thanks.

patrickdepinguin, 2018-01-19:
In case it helps, here are my kll files at this moment:

dvorak.kll (base layer)

(btw, not sure what changed with respect to BACKSPACE and BACKSLASH. Originally I had these lines uncommented, but with the update of controller and kll repos I did, this causes backspace to stop working. I had to comment these lines.)

```
Name = "MD1.1";
#Layout = "Alphabet";
#Base = "AlphabetBlank";
Version = "0.3d";
Author = "HaaTa (Jacob Alexander) 2014-2016";
KLL = "0.3c";
Date = "2016-02-06";
Generator = "KIICONF 0.2";

# Dvorak layout

U"ESC" : U"BACKTICK";
U"1" : U"1";
U"2" : U"2";
U"3" : U"3";
U"4" : U"4";
U"5" : U"5";
U"6" : U"6";
U"7" : U"7";
U"8" : U"8";
U"9" : U"9";
U"0" : U"0";
U"MINUS" : U"LBRACE";
U"EQUAL" : U"RBRACE";
#U"BACKSLASH" : U"BACKSPACE"; # copied from original qwerty, not sure what this is
U"TAB" : U"TAB";
U"Q" : U"QUOTE";
U"W" : U"COMMA";
U"E" : U"PERIOD";
U"R" : U"P";
U"T" : U"Y";
U"Y" : U"F";
U"U" : U"G";
U"I" : U"C";
U"O" : U"R";
U"P" : U"L";
U"LBRACE" : U"SLASH";
U"RBRACE" : U"EQUALS";
#U"BACKSPACE" : U"BACKSLASH";
U"CTRL" : U"LCTRL";         # capslock
U"A" : U"A";
U"S" : U"O";
U"D" : U"E";
U"F" : U"U";
U"G" : U"I";
U"H" : U"D";
U"J" : U"H";
U"K" : U"T";
U"L" : U"N";
U"SEMICOLON" : U"S";
U"QUOTE" : U"MINUS";
U"ENTER" : U"ENTER";
U"LSHIFT" : U"LSHIFT";
U"Z" : U"SEMICOLON";
U"X" : U"Q";
U"C" : U"J";
U"V" : U"K";
U"B" : U"X";
U"N" : U"B";
U"M" : U"M";
U"COMMA" : U"W";
U"PERIOD" : U"V";
U"SLASH" : U"Z";
U"RSHIFT" : U"RSHIFT";
U"FUNCTION2" : U"Function1";           # default: LCTRL
U"LALT" : U"LGUI";
U"LGUI" : U"LALT";
U"SPACE" : U"SPACE";
U"RGUI" : U"RALT";
U"RALT" : U"Function1";
U"FUNCTION3" : U"Function2";
U"FUNCTION4" : U"RCTRL";
```

fun1.kll
```
Name = "MD1.1";
#Layout = "Alphabet";
#Base = "AlphabetBlank";
Version = "0.3d";
Author = "HaaTa (Jacob Alexander) 2014-2016";
KLL = "0.3c";
Date = "2016-02-06";
Generator = "KIICONF 0.2";

U"ESC" : U"ESCAPE";
U"1" : U"F1";
U"2" : U"F2";
U"3" : U"F3";
U"4" : U"F4";
U"5" : U"F5";
U"6" : U"F6";
U"7" : U"F7";
U"8" : U"F8";
U"9" : U"F9";
U"0" : U"F10";
U"MINUS" : U"F11";
U"EQUAL" : U"F12";
U"BACKSPACE" : U"DELETE";

# left side arrows
U"E" : U"UP";
U"S" : U"LEFT";
U"D" : U"DOWN";
U"F" : U"RIGHT";

# right side arrows
U"I" : U"UP";
U"J" : U"LEFT";
U"K" : U"DOWN";
U"L" : U"RIGHT";

U"G" : U"INSERT"; # 'I' on Dvorak layout
U"X" : U"Lock3"; # lock switch to layer 3 ('Q' on Dvorak layout)

# Map 'alt' key of layer1 to layer2
U"LGUI" : U"Function2";
```

fun2.kll
```
Name = "MD1.1";
#Layout = "Alphabet";
#Base = "AlphabetBlank";
Version = "0.3d";
Author = "HaaTa (Jacob Alexander) 2014-2016";
KLL = "0.3c";
Date = "2016-02-06";
Generator = "KIICONF 0.2";

U"ESC" : flashMode();
U"1" : SYS"POWERDOWN";
U"3" : SYS"SLEEP";
U"4" : U"Lang3"; # Katakana, used for display settings (see xbindkeys)
U"MINUS" : CONS"VOLUMEDOWN";
U"EQUAL" : CONS"VOLUMEUP";
U"BACKSLASH" : CONS"MUTE";

# left side arrows
U"E" : U"PAGEUP";
U"S" : U"HOME";
U"D" : U"PAGEDOWN";
U"F" : U"END";

# right side arrows
U"I" : U"PAGEUP";
U"J" : U"HOME";
U"K" : U"PAGEDOWN";
U"L" : U"END";
```

qwerty.kll
```
Name = "MD1.1";
#Layout = "Alphabet";
#Base = "AlphabetBlank";
Version = "0.3d";
Author = "HaaTa (Jacob Alexander) 2014-2016";
KLL = "0.3c";
Date = "2016-02-06";
Generator = "KIICONF 0.2";

# Qwerty layout

U"MINUS" : U"MINUS";
U"EQUAL" : U"EQUAL";
#U"BACKSLASH" : U"BACKSPACE";
U"Q" : U"Q";
U"W" : U"W";
U"E" : U"E";
U"R" : U"R";
U"T" : U"T";
U"Y" : U"Y";
U"U" : U"U";
U"I" : U"I";
U"O" : U"O";
U"P" : U"P";
U"LBRACE" : U"LBRACE";
U"RBRACE" : U"RBRACE";
U"S" : U"S";
U"D" : U"D";
U"F" : U"F";
U"G" : U"G";
U"H" : U"H";
U"J" : U"J";
U"K" : U"K";
U"L" : U"L";
U"SEMICOLON" : U"SEMICOLON";
U"QUOTE" : U"QUOTE";
U"Z" : U"Z";
U"X" : U"X";
U"C" : U"C";
U"V" : U"V";
U"B" : U"B";
U"N" : U"N";
U"COMMA" : U"COMMA";
U"PERIOD" : U"PERIOD";
U"SLASH" : U"SLASH";
```

and my build script:
```
#!/usr/bin/env bash
#
# Keyboard: Infinity 60% with LED backlight support (MD1.1)
#
# These build scripts are just a convenience for configuring your keyboard (less daunting than CMake)
# Jacob Alexander 2016-2017

export PATH="/opt/sgpp-lite-arm-eabi-bin-2014.05.28-local/bin/:$PATH"

# Default to Alphabet
Layout=$(basename $0 | cut -d'.' -f2)
if [ "${Layout}" = "bash" ]; then
	Layout=alphabet
fi

#################
# Configuration #
#################

# Feel free to change the variables in this section to configure your keyboard

BuildPath="IC60_LED_custom"

# Define Layout Name
LayoutName=${Layout}

## KLL Configuration ##

# Generally shouldn't be changed, this will affect every layer
BaseMap="scancode_map scancode_map.${Layout}"

# This is the default layer of the keyboard
# NOTE: To combine kll files into a single layout, separate them by spaces
# e.g.  DefaultMap="mylayout mylayoutmod"
DefaultMap="custom/dvorak stdFuncMap"

# This is where you set the additional layers
# NOTE: Indexing starts at 1
# NOTE: Each new layer is another array entry
# e.g.  PartialMaps[1]="layer1 layer1mod"
#       PartialMaps[2]="layer2"
#       PartialMaps[3]="layer3"
PartialMaps[1]="custom/fun1 stdFuncMap"
PartialMaps[2]="custom/fun2"
PartialMaps[3]="custom/qwerty stdFuncMap"

##########################
# Advanced Configuration #
##########################

# Don't change the variables in this section unless you know what you're doing
# These are useful for completely custom keyboards
# NOTE: Changing any of these variables will require a force build to compile correctly

# Keyboard Module Configuration
ScanModule="Infinity_60_LED"
MacroModule="PixelMap"
OutputModule="USB"
DebugModule="full"

# Microcontroller
Chip="mk20dx128vlf5"

# Compiler Selection
Compiler="gcc"

########################
# Bash Library Include #
########################

# Shouldn't need to touch this section

# Check if the library can be found
if [ ! -f "${BASH_SOURCE%/*}/cmake.bash" ]; then
	echo "ERROR: Cannot find 'cmake.bash'"
	exit 1
fi

# Load the library
source "${BASH_SOURCE%/*}/cmake.bash"
```

patrickdepinguin, 2018-01-22:
Some extra info: when connecting to the CLI and enabling outputDebug, then typing the key for flashMode, there is no output at all. The CLI disconnects immediately. After that, the keyboard seems to reboot, and the tty is a new one (i.e. ttyACM0 is no longer there, becomes ttyACM1, later ttyACM2, etc.).

Using `picocom`, the last output is:
```
FATAL: term closed
```

In `dmesg`, I see:
```
[28699.234501] usb 3-1.4.4: USB disconnect, device number 7
[28699.234573] cdc_acm 3-1.4.4:1.4: urb 11 failed submission with -19
[28699.234619] cdc_acm 3-1.4.4:1.4: urb 12 failed submission with -19
[28699.234696] cdc_acm 3-1.4.4:1.4: urb 13 failed submission with -19
[28699.234776] cdc_acm 3-1.4.4:1.4: urb 14 failed submission with -19
[28699.234854] cdc_acm 3-1.4.4:1.4: urb 15 failed submission with -19
[28699.234935] cdc_acm 3-1.4.4:1.4: urb 0 failed submission with -19
[28699.235014] cdc_acm 3-1.4.4:1.4: urb 1 failed submission with -19
[28699.235094] cdc_acm 3-1.4.4:1.4: urb 2 failed submission with -19
[28699.235174] cdc_acm 3-1.4.4:1.4: urb 3 failed submission with -19
[28699.235254] cdc_acm 3-1.4.4:1.4: urb 4 failed submission with -19
[28699.235334] cdc_acm 3-1.4.4:1.4: urb 5 failed submission with -19
[28699.235414] cdc_acm 3-1.4.4:1.4: urb 6 failed submission with -19
[28699.235494] cdc_acm 3-1.4.4:1.4: urb 7 failed submission with -19
[28699.235574] cdc_acm 3-1.4.4:1.4: urb 8 failed submission with -19
[28699.235655] cdc_acm 3-1.4.4:1.4: urb 9 failed submission with -19
[28699.235738] cdc_acm 3-1.4.4:1.4: urb 10 failed submission with -19
[28699.298154] cdc_acm 3-1.4.4:1.3: acm_ctrl_irq - usb_submit_urb failed: -19
[28699.323050] cdc_acm 3-1.4.4:1.3: failed to set dtr/rts
[28699.526823] usb 3-1.4.4: new full-speed USB device number 8 using xhci_hcd
[28699.604781] usb 3-1.4.4: New USB device found, idVendor=1c11, idProduct=b04d
[28699.604786] usb 3-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28699.604789] usb 3-1.4.4: Product: Keyboard - Infinity_60_LED:alphabet PixelMap USB
[28699.604792] usb 3-1.4.4: Manufacturer: Input Club
[28699.604794] usb 3-1.4.4: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128
[28699.607367] input: Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:02.0/0000:3d:00.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4:1.0/0003:1C11:B04D.0016/input/input49
[28699.659167] hid-generic 0003:1C11:B04D.0016: input,hidraw2: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB] on usb-0000:3d:00.0-1.4.4/input0
[28699.660646] input: Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:02.0/0000:3d:00.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4:1.1/0003:1C11:B04D.0017/input/input50
[28699.712668] hid-generic 0003:1C11:B04D.0017: input,hidraw3: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB] on usb-0000:3d:00.0-1.4.4/input1
[28699.714259] input: Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:02.0/0000:3d:00.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4:1.2/0003:1C11:B04D.0018/input/input51
[28699.766487] hid-generic 0003:1C11:B04D.0018: input,hidraw4: USB HID v1.11 Device [Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB] on usb-0000:3d:00.0-1.4.4/input2
[28699.767002] cdc_acm 3-1.4.4:1.3: ttyACM0: USB ACM device
[28699.769201] hid-generic 0003:1C11:B04D.0019: hiddev97,hidraw5: USB HID v1.11 Device [Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB] on usb-0000:3d:00.0-1.4.4/input5
[28699.770624] input: Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:02.0/0000:3d:00.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4:1.6/0003:1C11:B04D.001A/input/input52
[28699.771019] hid-generic 0003:1C11:B04D.001A: input,hidraw6: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60_LED:alphabet PixelMap USB] on usb-0000:3d:00.0-1.4.4/input6
```

haata, 2018-01-22:
This might be related to an array bounds issue I fixed yesterday.
You may want to pull in the latest changes.

patrickdepinguin, 2018-01-22:
No, this did not change the behavior.

I have another issue, mentioning it here in case it could be related.
I notice that my second layer is not always working correctly anymore. Specifically, if I'm at the base layer, executing keys from the second layer work fine (when using layer shifting, that is). If however, I'm at layer 3, keys from the second layer do not work.

For example (referring to my earlier posts of my kll file):
* volume up, down, mute (defined in fun2.kll) only work when in base layer
* arrow keys (left-hand side) (defined in fun1.kll) work both from base layer as layer 3 (qwerty), but the Home/end/pgup/pgdown keys mapped on the same location as the arrows but in layer 2 (fun2.kll) do not work in layer 3.

Incidentally, the flashMode key is also mapped in Layer 2. However, it works in neither the base layer nor layer 3.

haata, 2018-02-27:
Found the issue while I was debugging another issue on the IC60.

https://github.com/kiibohd/controller/commit/ac024ca70826040d2a48c688232c52bed13fb828

Fixed now. Forgot to add an ifdef.

closed 2018-02-27 by haata