kiibohd/controller #272: Sequence Macro not outputting repeated characters

isabelacmor, 2018-01-21:
* **Device**:
 * WhiteFox TrueFox

* **Firmware Origin**:
 * Web config 

* **OS**:
 * Mac OS X, Windows 10

* **Reproduction Steps**:
From the sample [Sequence readme](https://github.com/kiibohd/controller/wiki/Layers,-Macros#sequences), you should be able to have the following in your KLL:
`U"F1" : 'Cool';` which outputs "Cool".

* **Resulting Issue**:
When triggered F1, the output is "Col", not "Cool". I tried with a variety of strings and no repeated characters were outputted. I also tried `U"F1" : U"C", U"O", U"O", U"L";` but that didn't work either.

haata, 2018-01-21:
I've had a few reports of this lately.

I'm starting to get the unit test validation working for sequence macros (combos are already working). Hoping to have something ready in the next few days.

This is how you'll be able to run the test using `controller/Keyboards/whitefox.truefox.bash`

```bash
EnableHostOnlyBuild=true HostTest=kll.py ./whitefox.truefox.bash
```

haata, 2018-01-30:
@isabelacmor while it's not up on the web configurator (or the desktop configurator) yet. I have reproduced (and hopefully fixed) the issue.

https://github.com/kiibohd/controller/commit/04e42b96cfaaece466837ff0d8301accecac43cb

haata, 2018-02-27:
Closing this, please re-open if it's not working.

zzachattack2, 2019-02-27:
> Closing this, please re-open if it's not working.

FYI this still seems to be an issue - compiling manually from docker with latest clone. Able to hack around it by sticking a dummy key between sequential repeated characters currently.

haata, 2019-02-27:
Interesting. Can you give me an example of the KLL expression you're using?

zzachattack2, 2019-02-27:
Sure, in the output directory I have a file "custom-white.kll" which is nothing except the following:
//
U"Q" : U"T", U"E", U"E", U"E", U"S", U"T";
//

The whitefox.bash script was modified with the following line:
//
PartialMaps[1]="whitefox/whitefox custom-white"
//

The result after flashing (just flashing the bin using the desktop application), when pressing Function+Q is "test"

If it helps, also using the WhiteFox TrueFox