kiibohd/controller #275: K-Type modifier keys either stuck or toggled randomly

mschwaig, 2018-02-13:
**Device**:
K-Type (Massdrop November 2017) with Halo switches

**Firmware Origin**:
lts release from web configurator with NoAnimations, flashed on a Ubuntu GNOME 16.04 LTS host as described in the docs

**OS**:
Issue occurs on Ubuntu GNOME 16.04 LTS, Windows 7 and Windows 10

**Version**:
I've attached the full output of `lsusb -d 1c11: -v` since I could not find the requested commit hash in the command output:
[lsusb_output.txt](https://github.com/kiibohd/controller/files/1721233/lsusb_output.txt)

**Reproduction Steps**:
Type normally but relatively rapidly on the keyboard, which generally works. The LEDs and Fn+Whatever commands for dimming and so on also work as expected. However frequently (every few keypresses-sometimes only every few minutes) the following issues suddenly appears:

**Resulting Issue**:
I notice that either Alt or Ctrl must have gotten triggered somehow so that Alt or Ctrl is now in a 'pressed state' and I completely mess up my inputs, closing windows and opening menus left and right.
I think this even happens when I did not even press a modifier key at all, but I am not certain since I never immediately notice it.

**Workaround**
I then have to stop typing and press all the modifier keys in order to get them back into the 'not pressed state' OR do the same thing on another keyboard that is also plugged in OR unplug the keyboard and plug it back in.
From what I read online think some people may also work around similar issues by using a 3.0 port without any USB HUB in between, which is not an option for me.

I can provide more info if required or help out if someone points me in the right direction. I'm a software developer with experience in C. I'm not sure this is a software issue though.

haata, 2018-12-15:
@mschwaig sorry for the really late reply (it's been a long year for me...). I'd recommend trying the latest firmware (the configurator will be updating in the next week or so with latest firmware).

I've fixed a lot of bugs in regards to stuck keypresses which could happen if exact timing is hit compared to the scan algorithm.

mschwaig, 2018-12-15:
@haata Thanks for the reply.

I will try out version 0.5.1 from a few days ago.
I saw there is also an updated binary available for the bootloader, not sure if I also have to update that.

mschwaig, 2018-12-15:
I have tried the 0.5.1 release, and the problem is still present on my model.
Running `showkey` I found that with this version the left Alt key stays stuck in the pressed state basically every time. Not sure if it could also happen with other keys.

mschwaig, 2018-12-16:
0.5.2 behaves the same for me.

I made a [video](http://photos.app.goo.gl/UhDAq5cUpDkdmaHh8) that shows the issue I am having. The Alt key stays stuck until I unplug the keyboard. After that I can type normally again on my laptop keyboard.

haata, 2018-12-16:
Yeah for your issue 0.5.2 probably wouldn't have fixed it.
Would you mind swapping the switch with a known working key and trying again? It might just be a faulty switch. If that's not it, there are more things to try.

Accessing the Virtual Serial Port, we can also use something similar to `showkey` directly on the keyboard to see if it's a hardware, firmware or USB issue.

https://kiibohd.github.io/wiki/#/Debugging

The particularly useful commands are `matrixDebug` and `outputDebug`. `matrixState` is useful to see if it's a hardware issue.

Also can you post me your `dmesg` output when you plug in the keyboard?

mschwaig, 2018-12-19:
Thanks for the help!

It looks like simply reseating the switch inside its socket fixed the issue for me.
I am a little bit embarrassed about that, but it was a great experience to have the person who designed the switch point that out to me, so thank you for that. :)

closed 2018-12-19 by mschwaig

haata, 2018-12-19:
:grinning: