kiibohd/controller #276: Sleep multimedia button doesn't work

vbuslov, 2018-02-17:
* **Device**:
K-Type (Massdrop November 2017)

* **Firmware Origin**:
web configurator

* **OS**:
Windows 10

* **Version**:
b1dc8310f4109561abb86ba7d72044f193787df9

* **Reproduction Steps**:
1. Configure f2+Backspace to 'Sleep'
```        {
            "code": "0x1F",
            "x": 52,
            "y": 6,
            "w": 8,
            "h": 4,
            "layers": {
                "0": {
                    "key": "BACKSPACE",
                    "label": "BKSP"
                },
                "1": {
                    "key": "DELETE",
                    "label": "DEL"
                },
                "2": {
                    "key": "SYS:SLEEP",
                    "label": "SLEEP"
                }
            }
        }
```
2. Press f2+Backspace.

* **Resulting Issue**:
Computer doesn't go to sleep, keyviewer doesn't register any keypresses.
[Here](https://github.com/qmk/qmk_firmware/blob/master/keyboards/dz60/keymaps/tailcall/keymap.c) is my dz60 keymap and KC_SLEP does work on same machine.

haata, 2018-03-15:
Would it be possible to try CONS:SLEEP instead?
The custom KLL would be:
```
U"Backspace" : CONS"Sleep";
```

It may be that Windows 10 doesn't like SYS:SLEEP anymore.

keybeejeebee, 2018-03-15:
not getting any response on that either (win 10). Just double checking that it's CON"Sleep" and not CONS:Sleep as you have written above?

haata, 2018-03-15:
I'll fire up a Windows box this weekend to test this.

PandorasFox, 2018-08-22:
The `SYS:POWER`, `SYS:SLEEP`, and `SYS:WAKEUP` keys don't do anything on windows - I'm unsure of exactly why.

`SYS:WAKEUP` _does_ correctly send `XF86WakeUp` to my linux rig, but the machine itself doesn't actually wake. I'm unsure if this is an issue with the BIOS on this rig doing something dumb to the USB buses or what, so I'll dig into that some more.

Mkenney11, 2019-02-13:
I am also not able to use the sleep key in the configurator to work.
Device: Kira (Kickstarter 2018)
Version: 0.5.4
OS: Windows 10 vs 1803

haata, 2019-02-18:
Should be fixed in the upcoming v0.5.5

closed 2019-02-18 by haata

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta

Download the K-Type bin file and use QuickFlash to load the firmware.
https://kiibohd.github.io/wiki/#/Configurator/Quickflash