kiibohd/controller #286: Ergodox Linux USB Issues with AMD X370 Chipset / ASMedia 1343 USB Hub

s3rb31, 2018-04-09:
**Devices**: 

* Infinity Ergodox Right-side Master (Massdrop 2017, USB-C Version)
* ASUS Prime X370-PRO Motherboard (with onboard ASMedia Device 1343 USB Hub)

**Firmware Origin**:  Both self-compiled and from the configurator (lts and latest) 

**OS**: archlinux with kernel 4.16.0-2-ARCH

**Version**:  bcdDevice 3.52

**Reproduction Steps**:

* (Boot with `xhci-pci.dyndbg=+pfl xhci-hcd.dyndbg=+pfl usbcore.dyndbg=+pfl`)
* Connect the keyboard to the computer 

Probably not reproducible without the mainboard / USB-controller in question.

**Resulting Issue**:

If connected to USB 3.1 Gen2 (Teal) ports there is an error message printed every 5 seconds: `Cannot enable. Maybe the USB cable is bad?`. I have attached a [dmesg output](https://github.com/kiibohd/controller/files/1891394/log-err.txt) for this case.

If connected to USB 3.1 Gen1 / USB3.0 (Blue) ports the error goes away but the device/port still hangs in a loop. Pretty much the same code loops except that in the function `hub_port_reset` the function `hub_port_warm_reset_required` returns false and thus the error message is [skipped](https://github.com/torvalds/linux/blob/master/drivers/usb/core/hub.c#L2839). I have attached another [dmesg output](https://github.com/kiibohd/controller/files/1891395/log-noerr.txt) output for this case.

* It definitely is not the cable, I tried multiple different cables 
* Even in flash-mode the error appears
* Other devices do not produce this or similar errors

**Additional Files**:

* [lspci.txt](https://github.com/kiibohd/controller/files/1891396/lspci.txt)
* [lsusb_ergo.txt](https://github.com/kiibohd/controller/files/1891398/lsusb_ergo.txt)

haata, 2019-01-30:
I've added a `usbErrors` to the virtual cli to see if there is a chipset issue.
We did notice for Kira that on some Ryzen motherboards the USB chipset (built-in one) was super flaky and produced a ton of USB protocol errors (plugged the same keyboard and cable into a different computer and there were no errors).

I would also check to see if more recent kernels have fixed some of the issues.

s3rb31, 2019-01-31:
Currently I am not using arch on the PC really, so I cannot check to see if the error still exists but I will check it out once I get a little bit of time to install arch.

I will have to update the ergodox firmware also to see if anything has changed. So far I have not felt the need to update it because it is quite stable, props to you there ;)