kiibohd/controller #291: Waking a suspending PC doesn't work

PandorasFox, 2018-08-23:
* **Device**: 
 Nightfox (Kono Store May 2018)

* **Firmware Origin**:
configurator

* **OS**:
Arch Linux, Windows 10

* **Version**:
3.55 (rev 853)

* **Reproduction Steps**:
  * Sleep the computer:
    * `sleep` in power menu on windows
    * `systemctl suspend` on linux
    * `SLEEP` key on keyboard, on linux (--> invokes `systemctl suspend` through logind on most systems)
  * Press keys on keyboard. Computer doesn't wake up.
  * Press keys on my connected POK3R. Computer wakes up.

* **Resulting Issue**:
I'm unable to wake up my computer without physically hitting the power button (pretty annoying, since it's out of reach). Other keyboards work fine and there's been issues made about this in the past, which seems to indicate it's a firmware bug. The build I'm on should include the last fix that was done for this, so it seems the last fix didn't fix it.

PandorasFox, 2018-08-23:
Apparently, waking up my PC works on the LTS firmware (version 2.02 / revision 514). Unsure what broke between those two commits; I might look into this if I have more time.

haata, 2018-12-15:
I've been doing some work with Suspend/Resume. Once the new configurator is launched (should be very soon), please let me know if there are still any issues.

haata, 2019-02-18:
I'm going to mark this as fixed, do re-open if you flash with the latest configurator and waking still doesn't work.

PandorasFox, 2019-04-05:
I just flashed today with the latest configurator - my pc won't wake on any presses from my nightfox.

PandorasFox, 2019-04-05:
Ah, never mind - I didn't realize that the web configurator isn't up to date. Just downloaded the appImage on arch and it worked like a charm - sorry about that!

closed 2019-04-05 by PandorasFox

haata, 2019-04-05:
@PandorasFox I'll take a look at this this evening. I thought this was working, but there have been a bunch of USB changes.