kiibohd/controller #297: Ergodox Infinty is non-functional since last commit

johngeorgewright, 2018-09-26:
* **Device**:

Infinity Ergodox Left-side Master (Massdrop 2016)

* **Firmware Origin**:

controller.git#master

* **OS**:

macOS High Sierra

**Version**:

Not 100% sure how to get this.

**Reproduction Steps**:

Fresh install of Ergodox Infinity and install bundled ergodox infinity binaries.
1. `git clone https://github.com/kiibohd/controller.git`
1. `docker build -f Dockerfiles/Dockerfile.ubuntu -t controller.ubuntu Dockerfiles`
1. `docker run -it --rm -v "$(pwd):/controller" controller.ubuntu`
1. `pipenv install && pipenv shell`
1. `./ergodox.bash`
1. Exit docker container and install binaries to ergodox

I should note that during the above steps I don't see any error or warning messages.

 **Resulting Issue**:

Logo appears but keyboard is non-functional until I install old binary (downloaded from https://configurator.input.club/).

**Solution**
1. git reset --hard HEAD~1

haata, 2018-09-26:
I'm assuming both sides aren't working.

This is likely to do with the SAM4S (Kira) merges we're doing right now. We'll be able to take a look at this in a week or two (should be an easy fix hopefully).
If you wouldn't mind, what is the last working git hash?

johngeorgewright, 2018-09-27:
No worries. I forked it and reverted that last merge. Working commit is currently `164a8f86004645a9ae36367c6a9044f577263d86`.

ganthern, 2018-11-10:
I'm still having this problem with commit 56cdf53. Was this supposed to be fixed and broke again or are we still "in a week or two"?

ganthern, 2018-11-11:
Just confirmed it with an up-to-date master branch: both sides were flashed with their respective firmware. Both sides are lighting up and showing logos, but no usb codes are sent (in left- and right master configuration).

haata, 2018-11-12:
Just pushed a small fix that helped on my Infinity Ergodox. I'm not done testing yet, but this should fix some of the problems.

haata, 2018-12-15:
I've done more fixes (https://github.com/kiibohd/controller/commit/e7b5587a220a11afd946dbab9dd89bc48bbd78d6) and have included some basic Animation support as well.

closed 2018-12-15 by haata