kiibohd/controller #301: The current arch linux dockerfile has incompatible pip and pipenv versions

ganthern, 2018-11-11:
* **Firmware Origin**:
github master branch

* **OS**:
Arch linux

* **Reproduction Steps**:
```
systemctl start docker

cd controller/Dockerfiles
docker build -f Dockerfile.archlinux -t controller.archlinux .
cd ..

docker run -it --rm -v "$(pwd):/controller" controller.archlinux

pipenv install && pipenv shell
```
* **Resulting Issue**:
```
[root@ea59b35642b3 Keyboards]# pipenv install && pipenv run
Creating a virtualenv for this project...
Pipfile: /controller/Keyboards/Pipfile
Using /usr/bin/python (3.7.0) to create virtualenv...
⠋Already using interpreter /usr/bin/python
Using base prefix '/usr'
/usr/lib/python3.7/site-packages/virtualenv.py:1041: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
New python executable in /root/.local/share/virtualenvs/Keyboards-41-guJFC/bin/python
Installing setuptools, pip, wheel...done.
Setting project for Keyboards-41-guJFC to /controller/Keyboards

Virtualenv location: /root/.local/share/virtualenvs/Keyboards-41-guJFC
Pipfile.lock (da0253) out of date, updating to (1ee067)...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
ils.py", line 402, in resolve_deps
    req_dir=req_dir
  File "/usr/lib/python3.7/site-packages/pipenv/utils.py", line 250, in actually_resolve_deps
    req = Requirement.from_line(dep)
  File "/usr/lib/python3.7/site-packages/pipenv/vendor/requirementslib/models/requirements.py", line 704, in from_line
    line, extras = _strip_extras(line)
TypeError: 'module' object is not callable
```

this seems related to https://github.com/pypa/pipenv/issues/2871 and https://github.com/pypa/pipenv/issues/2924 and can be fixed with
```
pip install pip==18.1 && pip install pipenv==2018.10.13
```

haata, 2018-11-12:
Try regenerating your archlinux docker image (may need to clear some of the cache). I think pipenv has fixed the issue now (I've been fighting this myself on Travis-CI and the kll repo).

haata, 2018-12-15:
Should be fixed now.

closed 2018-12-15 by haata