kiibohd/controller #302: K-Type keys not working randomly

manno, 2018-12-14:
* **Device**:

 * K-Type (Massdrop November 2017)

* **Firmware Origin**:

* compiled from master (controller+kll)

* **OS**:

* OSX

* **Reproduction Steps**:

```
cd Keyboards
pipenv run ./k-type.bash
dfu-util -D darwin18.K-Type.gcc.make/kiibohd.dfu.bin
```

* **Resulting Issue**:

Keyboard works with Linux, typing problems occur only with OSX:

* keys are ignored, only work on second or third press

haata, 2018-12-14:
Interesting.

@manno which version of macOS are you running?
And is this a MacBook? (if so which model?)

It's a bit unlikely, but it may also be the USB cable. I'm going to try this when I get home (I have a K-Type on my desk and a 13" 2017 MacBook Pro that I can test this on).

manno, 2018-12-14:
It's a 2016 MacBook Pro with 10.14.2. I tried three different cables.
Thanks!

haata, 2018-12-14:
I've confirmed this is an issue on my 2017 MacBook Pro (13"), macOS 10.14.1.

In addition to K-Type, this also affects WhiteFox, Ergodox and IC60.
Working on a fix now.

haata, 2018-12-15:
I'm not entirely certain the cause of the issue yet (likely some USB endpoint and interface mappings between the Kinetis and SAM4S MCUs) but this commit fixed the issue.

https://github.com/kiibohd/controller/commit/e7b5587a220a11afd946dbab9dd89bc48bbd78d6

Thanks so much for reporting this, took me most of the day to isolate the issue due to how odd the failure mode was. The keyboard wouldn't crash, the OS wouldn't initialize the keyboard so it would try to reset the keyboard and try again. It would be possible to type a few characters during this time so you could get some letters in, but then the port would be reset again. And so on.

When I do re-enable RawIO (currently unused), I'll get to the bottom of the mappings.

closed 2018-12-15 by haata