kiibohd/controller #303: Clear/reset function layer stack/ordering

ur5us, 2018-12-27:
* **Device**:
Infinity Ergodox Left-side Master (Massdrop 2016) [i.e. left side plugged into computer]

* **Firmware Origin**:
KiiboHD Configurator v0.4.1

* **OS**:
macOS Mojave v10.14.2

* **Reproduction Steps**:
`Lock-2`, `Next-f`, `Next-f`

* **Resulting Issue**:
Layer order will be 2 1 3

Cycling through all layers a few times usually clears the stack for me but it would be nice to trigger clearing on demand. Also related, though not sure how replicate: sometimes it would end being locked/stuck to particular layer and `Next-f` would stop working. The only way to reset is by unplugging.

haata, 2019-02-18:
This seems like a reasonable feature.

haata, 2019-02-18:
Added, will be included in the next firmware release.
May be a while before it's added as a configurator button, so it'll probably be easier to use custom KLL in the meantime once v0.5.5 reaches the configurator.

closed 2019-02-18 by haata