kiibohd/controller #310: Top LEDs Produce a High-Pitched Whine

GirthyMcGee, 2019-01-24:
* **Device**: Kira

* **Firmware Origin**: Kiibohd Configurator

* **OS**: Windows 10 Home v1809 OS build 17763.253

* **Version**: 1.0.0

* **Reproduction Steps**: Having top LEDs (under the keycaps) on

* **Resulting Issue**: High pitch whine coming from the motherboard. It's difficult to get a recording of it, but it really starts to get to me after about 10 minutes; similar pitch to when ears ring.

haata, 2019-01-30:
For those seeing this issue.

1. Try the keyboard with a different computer and/or different powered USB hub. Sometimes it's not the keyboard that is buzzing, but the device you've plugged into.
2. I will be making an animation mode that can isolate which LEDs are whining so it's easier to debug the issue. If it's a specific LED (or set of LEDs) then it's likely they are bad LEDs. However please try the 0.5.4 firmware when it's released as it may help the issue.

calfKillerRiver, 2019-02-01:
I'm experiencing this also, although I didn't notice it before this thread, so it's obviously not much of an issue for me. I'll help test if need be so we can confirm whether it's LEDs in the same location.

GirthyMcGee, 2019-02-02:
After doing some tests, the problem occurs over 3 computers I've tried (main computer, work computer, Surface). But, today I realized that you can adjust the brightness on these boards. I put my ear to the board and the whine seems to go away when it's really dim, and gets more and more pronounced the brighter the LEDs are.

haata, 2019-02-02:
I've added a few keys to help figure out the offending LEDs. This is only helpful if the sound is coming directly from an LED. If it's coming from the hub/computer/another part of the keyboard then you'll only hear it when more LEDs are on. Regardless this test is still useful in determining which category of LED whining you're in...

```
RCtrl+RShift+Esc => Go back to normal animations
RCtrl+RShift+Keypad 5 -> Reset test position
RCtrl+RShift+Keypad 6 -> Increment pixel LED (white)
RCtrl+RShift+Keypad 4 -> Decrement pixel LED (white)
RCtrl+RShift+Keypad 8 -> Increment LED channel (Red, Green or Blue)
RCtrl+RShift+Keypad 2 -> Decrement LED channel (Red, Green or Blue)
```

The increments may not seem in order, but they are in order by LED driver chip channel number and Pixel number (LED numbers printed on the back of the pcb).

Now that this test is in, I'm going to move this bug to the next release as I don't have a complete fix yet (still need to collect more information).

schnikes, 2019-02-07:
> After doing some tests, the problem occurs over 3 computers I've tried (main computer, work computer, Surface). But, today I realized that you can adjust the brightness on these boards. I put my ear [...]

I am getting the exact same issue. Flashed w/current firmware. The high pitch noise is 100% coming from the board and not any other device.

calfKillerRiver, 2019-02-15:
Hi haata. I just used your testing tool and here's what I'm able to figure out. I can't hear it at all from any single LEDs from the testing tool. I think that means it's not happening, but I'm unwilling to definitely assert that - it could be that from one LED the volume is too low. When I page through different LED animations and colors (where all keys are lit), I can always hear it - it doesn't seem to be color-dependent or animation dependent. However, you know how when you hold down the macro keys, that keys with functions in that layer pulse? When I do that, the sound pulses. (I did go through again after noticing this with the testing tool, paying special attention to any pulsing keys, but still can't hear it from any single LED). I'm suspecting that this is more complicated that faulty LEDs because Girthy mentioned "the top row", which is where mine is coming from, too. Specifically, I think maybe it's loudest approximately behind F12 - is there a capacitor back there?

haata, 2019-02-17:
@calfKillerRiver there isn't a capacitor there, so it's almost certainly coming from an LED (there are some caps closer to the bottom of the backspace and back slash key though).

If using a powered USB hub doesn't help you, I'd recommend contacting support@kono.store

stoikos, 2019-02-23:
I'm probably too stupid for this but "RCtrl+RShift+Esc" does not enable normal animations, it enables flash mode. How am I supposed to do the test while in flash mode? What am I missing? Just to make clear: The sounds is definitely coming from the LEDs on mine.

haata, 2019-02-23:
@stoikos contact support@kono.store if you think the buzzing is coming from the LEDs (putting your ear over the LEDs is a pretty good way to determine), they'll sort you out.