kiibohd/controller #312: Single color animations do not save

haata, 2019-01-28:
* **Device**:

Kira

* **Firmware Origin**:

controller.git

* **OS**:

Any (Linux)

* **Version**:

0.5.3

* **Reproduction Steps**:

* Select a single color animation
* Save the animation
* Restart keyboard

* **Resulting Issue**:

* Animation saved does not come back

(This is due to the position of the animation not being saved as static color selection is done by using a frame within an animation)

haata, 2019-02-04:
Fixed! Took more effort than I was hoping for...

closed 2019-02-04 by haata

mattkemp, 2019-02-20:
This is still an issue for me, using the orange animation (and changing it to teal) along with fingerprints_two_tone. Saving doesn't seem to do anything (all LEDs go dark on key press), it also loses the blue when unplug and re-plug. The fingerprints still work though.