kiibohd/controller #313: Lock LEDs not working

haata, 2019-01-28:
* **Device**:

Kira

* **Firmware Origin**:

controller.git

* **OS**:

Windows (though on macOS and Linux to some extent)

* **Version**:

0.5.3

* **Reproduction Steps**:

* Try CapsLock and NumLock keys

* **Resulting Issue**:

* Doesn't turn on the lock LEDs (sometimes, or always)

haata, 2019-01-29:
Should be fixed with https://github.com/kiibohd/controller/commit/15a37dcc529d399ef1559e3b8e95ff2edcc1329f

haata, 2019-01-31:
Re-opening, still some reports of lock leds not working with the latest firmware.

haata, 2019-02-01:
Should be fixed this time. If anyone still has issues after v0.5.4 let me know and I can provide some debug firmware to help figure out what's going on.

rondog, 2019-02-05:
I just flashed my kira with the 0.5.4.bin and the LEDs still do not illuminate on my mac. I used the configurator tool to do the flashing and used this file: https://github.com/kiibohd/controller/releases/download/v0.5.4/kira.v0.5.4.dfu.bin

haata, 2019-02-05:
@rondog which mac do you have?

I just tried it on my Macbook Pro (13" 2017) and only the CapsLock seems to work. But it's interesting because you have to hold it down for a second or so. Caps Lock isn't activated until the LED turns on (the laptop keyboard does the same thing).

rondog, 2019-02-05:
@haata I have a 27" iMac, Late 2013. My capslock will activate immediately when I press it. The press and hold sounds like its related to the macbook specifically. I tried holding down my caps lock, but that didn't make a difference. Capslock still works, just no LED to indicate if its on or not

haata, 2019-02-05:
Thanks @rondog, I'm going to see if I can test on an older Mac Mini to see if that makes any difference.
Also, to make sure you're not using Karabiner or USB Overdrive?

I'm going to see if I can build a test firmware where we can see if something else is going wrong (maybe add a cli command to enable lock led debugging).

rondog, 2019-02-05:
oh crap I actually have been using karabiner with my last keyboard. I just uninstalled it and restarted my computer, but same issue persists. I think I was only remapping the F keys with karabiner anyway. Also, not using USB overdrive. I'd be happy to assist with any debugging.

haata, 2019-02-06:
@rondog thanks!
Ok, so the next thing to try is to enabled 6KRO mode on the keyboard (RCtrl+RShift+F3), then try CapsLock again.

I've semi confirmed that macOS doesn't really have a NumLock or ScrollLock function so those LEDs won't be doing anything (at least until I add animation support for them).

If that doesn't work I'll link you a file to flash on your Kira that adds a debug mode which will tell us what (if anything) the OS is telling us about CapsLock. The way lock LEDs work is that you press the key, the OS activates that function and then the OS tells all the connected keyboards that they should turn on the lock LED for that function.

rondog, 2019-02-06:
@haata I am not seeing any difference in 6KRO mode.

haata, 2019-02-18:
I did more work on this with 3ba7436d86a81bca7dd69b467d78c7be6591fe2b so there may be some improvement for the upcoming v0.5.5

haata, 2019-02-24:
Closing this issue, once v0.5.5 is released, please verify again.

arcrox, 2019-03-18:
Lock LEDs are (still) non-functional on v0.5.5 for me (Win 10 x64).

haata, 2019-03-18:
I'll need to test this again it seems.
It's possible the keyboard needs a power cycle after flashing (uncommon, usually just bad luck) where the LED controller isn't fully working. But if animations are working, this wouldn't be the case.

arcrox, 2019-03-18:
No, but My USB controller says Intel USB 3.1 eXtensible Host Controller 1.10, driver version 10.0.14393.1794.  Computer is an HP Elitedesk enterprise mini workstation.  Power cycling doesn't work unfortunately, and I've played with different USB ports and removing potential conflicting devices and different cables.

haata, 2019-03-19:
@arcrox Would you mind doing a little debugging for me? (with the v0.5.5-beta firmware)

1. Try switching keyboard to 6KRO mode (RCtrl+RShift+F2). Then try CapsLock and NumLock keys to see if either of the indicators work.
2. Then switch back to NKRO mode (RCtrl+RShift+F3) and try again.
3. Open up the virtual serial port (this takes a bit of setup https://kiibohd.github.io/wiki/#/Debugging) and enter the `outputDebug` command. This will print out the USB keyboard packets being sent by the keyboard and any packets the OS sends the keyboard regarding lock LEDs. Press NumLock a few times and send me the output.
This will help me figure out where the issue is.

As a side note I quickly tested on a X230 with Windows 10 x64 last night as my lock LEDs are working. However, I have noticed that some Windows systems handle lock leds differently for some reason.

arcrox, 2019-03-20:
Steps 1-2 to switch between 6KRO and NKRO did not result in the indicators working.
For step 3 (virtual serial port), I'd be happy to, but I'm afraid I'll need the "instructions for dummies" treatment.  I'm attempting to install Cygwin: what package(s) do I install, and what do I do from there to get to the terminal to input the outputDebug command?

On a side note, I also tested this on a laptop (HP EliteBook x360 1030 G3) with the same results.

haata, 2019-03-22:
@arcrox are you sure you're using the latest beta firmware?

I updated the wiki I linked, it only includes the putty instructions now (much easier to use than cygwin). The trick on windows is figuring out which COM port to use (Device manager can give you some hints). We're working on a way to have the serial console available from the configurator (but it's a little ways out).

I think I move this issue out to v0.5.x so I can release the current version (Kinetis based keyboards have broken LEDs right now).

arcrox, 2019-03-22:
Yes, I flashed kira.v0.5.5-beta.dfu.bin using the Kiibohd configurator.

Here is the output from Putty (I pressed capslock, then numlock):

[image]

Thank you!