kiibohd/controller #315: Kira - Keys not outputting after reassign and flash

sayHIGHtome, 2019-01-30:
* **Device**: Kira (Kickstarter 2018)
* **OS**: Windows 10 Pro

* **Reproduction Steps**: 
Open configurator; select kira keyboard, select PGUP key on base layer, click reassign button, select "; :" key; repeat for "PGDWN" to reassign to "BKSP"
Click download firmware button, place keyboard into flashing made, click button to flash.
Keyboard is then flashed, screen is returned back to layout of keyboard

* **Resulting Issue**: 
When using keys, it doesn't do anything.

haata, 2019-01-30:
Would you mind attaching the JSON files?

sayHIGHtome, 2019-01-30:
Is there an easy way to export it from the configurator?
Or do i have to paste it into notepad or something?

I went back in and played around (put BKSP on something else to see if it worked there) and found that it's just my two keys (PGUP, PGDWN) that aren't working.