kiibohd/controller #316: LED Indicator Activate / Deactivate not triggering macro

cathysax, 2019-01-31:
* **Device**: Kira, from Kono Store, received January 2019
* **Firmware Origin**: kira-firmware-v0.5.3-installer.exe
* **Kira Firmware Version**: v0.5.3 (+0:1103)
* **OS**: Windows 10 Pro (x64)

* **Reproduction Steps**:
  * Put the following in the Custom KLL section of the Configurator:
  ```
  I"NumLock"(A) : A[Numpad_Layout](start);
  I"NumLock"(D) : A[Main_Layout](start);
  ```
  * Add the two animations (mine are single-frame animations)
  * Compile/flash to Kira keyboard
  * Press NumLock key

* **Resulting Issue**: The desired animation is not loaded. It should change the lights under the numpad keys based on whether NumLock is enabled (using the state of the LED).

* **Additional Information**: 
  * The NumLock LED on the side of the keyboard does toggle as expected.
  * The `A[...](start)` code works as expected when assigned to a key.
  * `I"NumLock" : A[Numpad_Layout](start);` (without (A) or (D)) is recognized correctly as a trigger, but of course always sets to the numpad lighting, even when NumLock is toggled off.
  * The (A) and (D) are activate / deactivate states documented in KLL spec v0.5c (page 11, section 5.1.4[.1]).
  * I'm using the default Win 10 HID keyboard driver since the the Lock indicator LEDs didn't work with the Kira-specific driver.

haata, 2019-02-01:
Thanks for opening this, this functionality is **almost** working (though I have some pieces disabled atm).
I'm planning on getting this into the v0.6.0 release in a few weeks (though I may have a beta release much sooner than that).

cathysax, 2019-11-28:
Do you have an expected timeframe when this will be ready? I'd also like this functionality for the caps lock key. Hopefully that's an easy / equivalent modification of code. Thanks!