kiibohd/controller #319: Compilation Failed Error on Any Change To the Standard Layout

uphoric, 2019-02-01:
* **Device**:
Kira

* **Firmware Origin**:
git/Kira online manual

* **OS**:
Windows 10 64bit

* **Version**:
Version 1.0

* **Reproduction Steps**:
I followed the steps outlined on this page:

https://kiibohd.github.io/wiki/#/Keyboards/Kira

>> Downloaded the drivers from the link
>> Ran the install
>> Plugged in the keyboard
>> Put it in flash mode
>> Installed the drivers
>> Donloaded Kiibohd desktop application
>> Ran the desktop application
>> Changed any key or LED assignment on any layer

* **Resulting Issue**:
Receive a "Compliation Failed" error.  When I press the "show log" button in an attempt to debug, nothing appears.

Interestingly enough, if I don't change anything on the standard layout, I am able to flash the keyboard.  I am also able to extract the .json, edit, upload, and flash.  Editing the key assignments or LED's via the GUI results in an error.

uphoric, 2019-02-01:
As a follow up, I was able to get a an error log by removing some of the animation profiles (miami_wave, kira_wave, etc.).  After removing the profiles I tried to compile and received a Compliation Failed error.  I then clicked on show log and received the following:

linux-gnu/Linux is supported.
Unix Build
Selected Generator: Ninja
/tmp/tmp.WGF9gVvbUl
VID: 0x308F (Boot: 0x308F)
PID: 0x0013 (Boot: 0x0012)
-- Build CPU Detected:
x86_64
-- Build Kernel Detected:
Linux 4.15.13-x86_64-linode106
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
sam4s8b
-- Chip Family:
sam4s
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/sam.c;Lib/delay.c;Lib/entropy.c;Lib/gpio.c;Lib/periodic.c;Lib/storage.c;Lib/time.c;Lib/arm_cortex.c;Lib/ASF/common/services/clock/sam4s/sysclk.c;Lib/ASF/common/utils/interrupt/interrupt_sam_nvic.c;Lib/ASF/sam/drivers/efc/efc.c;Lib/ASF/sam/drivers/pmc/pmc.c;Lib/ASF/sam/utils/cmsis/sam4s/source/templates/system_sam4s.c;Lib/ASF/sam/services/flash_efc/flash_efc.c
-- Bootloader Type:
dfu
-- The C compiler identification is GNU 6.3.1
-- Check for working C compiler: /usr/lib/ccache/arm-none-eabi-gcc
-- Check for working C compiler: /usr/lib/ccache/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Full Compiler:
/usr/lib/ccache/arm-none-eabi-gcc 6.3.1
-- Detected Scan Module Source Files:
Scan/Devices/ISSILed/i2c.c;Scan/Devices/ISSILed/led_scan.c;Scan/Devices/MatrixARMPeriodic/matrix_scan.c;Scan/Devices/Storage/storage.c;Scan/Kira/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/kll.c;Macro/PartialMap/layer.c;Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c;Macro/PixelMap/pixel.c;/tmp/tmp.WGF9gVvbUl/generatedPixelmap.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/Interface/output_gen.c;Output/USB/output_com.c;Output/USB/output_usb.c;Output/USB/arm/usb_desc.c;Output/USB/arm/usb_dev.c;Output/USB/arm/usb_joystick.c;Output/USB/arm/usb_keyboard.c;Output/USB/arm/usb_mem.c;Output/USB/arm/usb_mouse.c;Output/USB/arm/usb_rawio.c;Output/USB/arm/usb_serial.c;Output/USB/../../Lib/ASF/sam/drivers/udp/udp_device.c;Output/USB/../../Lib/ASF/common/services/usb/udc/udc.c;Output/USB/../../Lib/ASF/common/services/usb/class/cdc/device/udi_cdc.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/latency/latency.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Found LSB_RELEASE: /usr/bin/lsb_release  
-- Build OS Detected:
Ubuntu 18.04.1 LTS bionic
-- Found Ctags: /usr/bin/ctags-exuberant (found version "5.9~svn20110310") 
-- Found PythonInterp: python3 (found suitable version "3.6.7", minimum required is "3") 
-- Checking for kll
-- GITHUB_APIKEY was set.
-- kll Version Detected: kll 0.5.7.3
-- kll Installation Path: /root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll
-- kll Layouts Cache Path: /tmp/7-hid-io-layouts-c5410a1
-- Detected Layout Files:
/controller/Scan/Devices/ISSILed/capabilities.kll
/controller/Scan/Devices/MatrixARMPeriodic/capabilities.kll
/controller/Scan/Devices/Storage/capabilities.kll
/controller/Macro/PartialMap/capabilities.kll
/controller/Macro/PixelMap/capabilities.kll
/controller/Output/HID-IO/capabilities.kll
/controller/Output/USB/capabilities.kll
/controller/Debug/latency/capabilities.kll
/controller/Debug/led/capabilities.kll
/controller/Scan/Kira/scancode_map.kll
/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/layouts/stdFuncMap.kll
/tmp/tmp.WGF9gVvbUl/Kira-Standard-0.kll
/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/layouts/stdFuncMap.kll
/tmp/tmp.WGF9gVvbUl/Kira-Standard-1.kll
/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/layouts/stdFuncMap.kll
[... 70 lines cut ...]
                      ^~~~~~~~~~~~~~~~~~~~~
generatedPixelmap.c:400:22: error: 'Animation__orange_fade_in' undeclared here (not in a function)
  { (TriggerMacro*)0, Animation__orange_fade_in, /*23 A[orange_fade_in](replace:clear,start)*/
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
generatedPixelmap.c:402:22: error: 'Animation__pastel_wave' undeclared here (not in a function)
  { (TriggerMacro*)0, Animation__pastel_wave, /*24 A[pastel_wave](replace:clear,start)*/
                      ^~~~~~~~~~~~~~~~~~~~~~
generatedPixelmap.c:404:22: error: 'Animation__rainbow_wave' undeclared here (not in a function)
  { (TriggerMacro*)0, Animation__rainbow_wave, /*25 A[rainbow_wave](replace:clear,start)*/
                      ^~~~~~~~~~~~~~~~~~~~~~~
generatedPixelmap.c:406:22: error: 'Animation__test_animation' undeclared here (not in a function)
  { (TriggerMacro*)0, Animation__test_animation, /*26 A[test_animation](single)*/
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
generatedPixelmap.c:408:22: error: 'Animation__underlighting_only_rotation' undeclared here (not in a function)
  { (TriggerMacro*)0, Animation__underlighting_only_rotation, /*27 A[underlighting_only_rotation](pos:0,single)*/
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[24/50] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/layer.c.obj
[25/50] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/result.c.obj
[26/50] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/trigger.c.obj
[27/50] Building C object CMakeFiles/kiibohd.dir/Output/Interface/output_gen.c.obj
[28/50] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/macro.c.obj
[29/50] Building C object CMakeFiles/kiibohd.dir/Output/HID-IO/hidio_com.c.obj
[30/50] Building C object CMakeFiles/kiibohd.dir/Macro/PixelMap/pixel.c.obj
ninja: build stopped: subcommand failed.
Error in build. Exiting...

calfKillerRiver, 2019-02-01:
I'm experiencing this also, on macos, although it's not for _any_ change, just lots of them. (e.g. I was able to swap some alts/GUIs/CTRL keys on the base layer to match my laptop layout, but almost any change to layer 2 would result in a fail to compile.

romedtino, 2019-02-09:
A bit of a work around that's worked for me so far if all you want to do is remove assigned key values. 

1. Switch to a compiling layout
2. Grab the JSON file
3. Paste in your favorite editor and for every key you want to remove the assignment
   1. Search for the key name by label e.g. `FILES`
   2. Remove it on the layer you don't want it on. For example:
 ```
"layers": {
        "0": {
          "key": "F",
          "label": "F"
        },
        "1": {
          "key": "CONS:FILE BROWSER",
          "label": "FILES"
        }
      }
```
becomes
```
"layers": {
        "0": {
          "key": "F",
          "label": "F"
        }
      }
```
The key is to not forget the `,` after the "0" key value

Rinse and repeat. 

I think kiibohd is mainly GUI related? but I just got my Kira and was really confused why just removing the "WEB" media shortcut on my layer 1 would make it compile. There's a bug somewhere in the code and hopefully this helps people workaround it. GL!

jaredmixpanel, 2019-02-13:
Download Firmware results in "Compilation Failed" 100% of the time for me. Even with no changes at all to the default standard layout. Also clicking the "SHOW LOG" button does nothing.

Kira firmware v0.5.4, macOS 10.14.2, Kiibohd 1.0

[build.log](https://github.com/kiibohd/controller/files/2861690/build.log)