kiibohd/controller #321: Kira becomes unusable after creating fingerprints animation

thousand, 2019-02-03:
* **Device**: Kira

* **Firmware Origin**: Installed recommended firmware via updater (https://github.com/kiibohd/kiidrv/releases/download/v1.5.3-kiidrv/KiraFirmwareInstaller.app.zip), configutation done with kiibohd configurator, mac edition (https://github.com/kiibohd/configurator/releases/download/v1.0.0/kiibohd-configurator-1.0.0-mac.dmg)

* **OS**: macOS 10.14.2

* **Version**: (used `lsusb` installed via `homebrew`, FYI) 
```
$ lsusb -d 1c11 -v
2019-02-03 11:03:19.158 system_profiler[86571:10423665] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCISPT
      PCI Device ID: 0xa12f
      PCI Revision ID: 0x0031
      PCI Vendor ID: 0x8086

        Keyboard - Kira PixelMap USB:

          Product ID: 0x0013
          Vendor ID: 0x308f
          Version: 4.4f
          Serial Number: 5338310031504C353330303234353030 - sam4s8
          Speed: Up to 12 Mb/sec
          Manufacturer: Input Club
          Location ID: 0x14600000 / 14
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

        iBridge:

          Product ID: 0x8600
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 1.01
          Manufacturer: Apple Inc.
          Location ID: 0x14200000

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCIAR
      PCI Device ID: 0x15d4
      PCI Revision ID: 0x0002
      PCI Vendor ID: 0x8086
      Bus Number: 0x00

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCIAR
      PCI Device ID: 0x15d4
      PCI Revision ID: 0x0002
      PCI Vendor ID: 0x8086
      Bus Number: 0x01
```

* **Reproduction Steps**:
In configurator:
* With "Show Visuals" mode active
* Activate the "Customize Prebuilt Animation" tab
* Select `fingerprints` or `fingerprints_two_tone` from "Animations" dropdown.
* Select any color(s) from color picker(s)
* Click "Create" button
* Activate the "Animations Overview" tab
* Select newly created animation as the only "Startup Animation"
* Click "Download Firmware" button, follow firmware update procedure for Kira hardware

* **Resulting Issue**:
* No keys work
* Lights on keyboard are frozen from previous state, but repeatedly and briefly turn off in about 2/3 second cycle (all lights off for about 1/10th second per cycle)

* **Other Notes**:
* Correct behavior can be restored by deleting newly created "fingerprints" animation from "Animations Overview" tab and applying that configuration.

stoikos, 2019-02-12:
Could you elaborate on that? How do we enable fingerprints otherwise if not as a startup?

thousand, 2019-02-12:
Agreed, the intent is not clear in the configurator. I think it's working as intended with it added but not enabled as the startup animation.

keybeejeebee, 2019-02-12:
The fingerprints animation is reactive, not a traditional animation. When you add the animation in the configurator, it adds custom kll to your config that tells the keyboard to play the animation each time a key is pressed. You can see the custom kll by hiding the visuals tab, and finding the custom kll tab in the lower left (it may have a different name in the new version, going by memory at current).

gromwalh, 2019-02-12:
I think the issue is that some of us just want the fingerprint animation, and it seems odd that we can't just set that as the default when the keyboard starts up (without having to press some other key combination to activate it).  Is there a way to let us to that (even if it's custom kll that we need to add), or is the current firmware not capable of this?

keybeejeebee, 2019-02-12:
ok, try this. remove or deactivate all other animations besides the fingerprints and flash your board.

Type anything and you will see fingerprint animations. 

You don't need to activate or set as default, it just works.

stoikos, 2019-02-12:
That never works for me unfortunately. It always renders the keyboard completely useless - I get either crazy flashing LEDs or some other type of malfunction, most of the time it won't even register typing or even mess with my mouse for some reason (mouse getting very laggy). So unless I delete fingerprint and re-download the firmware I'm pretty much left with a non usable keyboard (and mouse) whenever I try this.

nick-solly, 2019-02-12:
I'm experiencing exactly the same issue.

keybeejeebee, 2019-02-12:
are you setting the fingerprints animation as startup? You should not do that, leave it unchecked, slid to the left, grey, whatever.

stoikos, 2019-02-12:
No, everything is unchecked. I tried the startup thing as well as the proper way. Nothing works. It also makes no difference if it's the regular 'fingerprint' or 'two-tone-fingerprint'. I should say I also have a broken LED, not sure if that affects 'fingerprint' (it does not affect any other animation though).

nick-solly, 2019-02-12:
Also have tried the above suggestions.  I don't have a broken LED but still have the "fingerprint" issue.

timokoesters, 2019-02-22:
I think the issue only arises when the name of the fingerprints animation is "fingerprints" as well. Try to change the name and see if it works.

stoikos, 2019-02-22:
I think I initially tried changing the name in the beginning and it didn't work. Now that I tried again, it sort of works, but only to a degree - meaning that it actually registers the fingerprint animation but the problem is that all LEDs are set to a pinkish colour until I actually start pressing keys. When I press a key it switches to the specific colour or two-tone that I chose for the fingerprint animation. But all other keys that haven't been pressed yet still remain pink. Also, 4 specific keys don not register the change at all, and just remain pink (two arrow keys and 'insert' and 'delete'). So it's not really working. And no, I'm not using it as a start-up animation. This is on a Mac btw, haven't tried fingerprint on Windows yet. I might do that over the weekend and see if there's a difference.

mattkemp, 2019-02-25:
I raised the four keys not working as a configurator issue as well https://github.com/kiibohd/configurator/issues/126

haata, 2019-03-14:
The four keys should be fixed now https://github.com/smasher816/kiisrv/commit/432908a890e16771dbe0404224ac994748a35344

haata, 2019-03-15:
I need to think more about how to make reactive animations "default". They are triggered in a fundamentally different way, so this will involve a combination of KLL, firmware and the configurator.

stoikos, 2019-03-26:
Has the fix for the 4 keys already been implemented? I updated the firmware, but no change on my end. Other than that, it seems to be working fine.

mattkemp, 2019-05-30:
I don't think the four key fix has made it into the latest firmware yet? Updated again today and still not working.

chasdevs, 2020-05-04:
Also hit this same issue today.

Herkkupala, 2020-05-04:
I tried to paste the code over to kira_wave instead of a complete new animation, it just won't play the animation but the keyboard won't brick either. Instead it would show a static version of the previous animation.