kiibohd/controller #322: Kira Num,Caps,Scroll indicators does not work on my instance of Windows 10

kolivusha, 2019-02-04:
Num,Caps,Scroll indicators (indicators) are OFF on my board. For example Caps Lock indicator does not react to the Caps Lock press, even though it is functional and switching it in the OS, confirmed with TYPING and audio ques (accessibility settings).
However indicators are working just fine on installs of windows, (same PC, different OS), and resetting my windows did not solved the problem, it also works on other laptop. Both systems were Indicators worked were plug and play, they did not saw any drivers for this board.

What could be the solution?

my os: Windows 10 Pro 64-bit (10.0, Build 17134) (17134.rs4_release.180410-1804)
Configuration v1.0.0 (default firmware flashed)
kira-firmware-v0.5.3

haata, 2019-02-04:
The new firmware update (https://github.com/kiibohd/controller/releases/tag/v0.5.4) fixes the issue.
It's still not live in the configurator yet, so it'll be a day or two longer.

kolivusha, 2019-02-04:
Replaced "kiibohd.dfu.bin"  with "kira.v0.5.4.dfu.bin" (renamed to the same name) in kira-firmware-v0.5.3-installer.exe (opened with 7zip). Flashed my Kira ==> Successes!!! 
Problem gone!
Thanks a lot! 

Modified file attached.
[kira-firmware-v0.5.3-installer modiefied to 0.5.4 .exe.zip](https://github.com/kiibohd/controller/files/2830080/kira-firmware-v0.5.3-installer.modiefied.to.0.5.4.exe.zip)

closed 2019-02-04 by haata