kiibohd/controller #323: Request: option to disable fade / breathe (Kira)

cathysax, 2019-02-05:
* **Device**: Kira, from Kono Store, received January 2019

* **Firmware Origin**: kira-firmware-v0.5.3-installer.exe

* **Kira Firmware Version**: v0.5.3 (+0:1103)

* **OS**: Windows 10 Pro (x64)

* **Reproduction Steps**: Create a single-frame animation and set it as the default (e.g. using Configurator)

* **Resulting Issue**: The LED brightness levels increase and decrease. This is documented as a feature ("breathing"), but I'd like a way to disable the fade and just keep the LED levels I set. Thanks for considering this.

haata, 2019-02-06:
Sounds reasonable, I'll queue it up for the next release as it shouldn't be hard.
Basically a key that resets/turns off fade for keys, underlighting and lock leds (which is set to fade profile 0).

cathysax, 2019-02-06:
Thanks for scheduling this feature.
When I hit the key to disable fade, would it revert to my specified LED levels, or keep it at whatever it was when I hit the disable key?
Ideally, I'd like to be able to set the fade profile default (to 0) in the firmware (e.g. like I set the startup animation) so that I don't have to disable fade every time I boot my computer.

shadowbottle, 2019-02-12:
Was this applied in the most recent firmware? Breathe is supposed to be by default, but I have to manually set timing to enable breathe at all.

haata, 2019-02-24:
Added more fade control variables and capabilities.
ae7bf0b711dfba5e77451cd2f269032f929a8657

https://github.com/kiibohd/controller/blob/ae7bf0b711dfba5e77451cd2f269032f929a8657/Macro/PixelMap/capabilities.kll#L89
and
https://github.com/kiibohd/controller/blob/ae7bf0b711dfba5e77451cd2f269032f929a8657/Macro/PixelMap/capabilities.kll#L102

closed 2019-02-24 by haata