kiibohd/controller #324: Kira keyboard media keys stopped working after mandatory firmware flash

kajotac, 2019-02-06:
* **Device**: Kira (Kono Store)

* **Firmware Origin**: mandatory firmware update (0.5.3)

* **OS**: Linux (debian and arch linux)

* **Version**: bcdDevice 4.4f

* **Reproduction Steps**: Pressing the Right Shift + Right Control + Media Key blinks the lights and does not do what's supposed to do (volume up, volume down or play)

* **Resulting Issue**: Lights start to blink and media combination does not do what's programmed to do (vol up, vol down, play don't work)

haata, 2019-02-06:
@kajotac Can you post the results of xev when pressing each of those keys?

(I'll test later this evening on Arch to make sure nothing is broken as well)

kajotac, 2019-02-06:
mrk@big-nuc:~$ xev
Outer window is 0x4a00001, inner window is 0x4a00002

PropertyNotify event, serial 8, synthetic NO, window 0x4a00001,
    atom 0x27 (WM_NAME), time 326861, state PropertyNewValue

PropertyNotify event, serial 9, synthetic NO, window 0x4a00001,
    atom 0x22 (WM_COMMAND), time 326861, state PropertyNewValue

PropertyNotify event, serial 10, synthetic NO, window 0x4a00001,
    atom 0x28 (WM_NORMAL_HINTS), time 326861, state PropertyNewValue

CreateNotify event, serial 11, synthetic NO, window 0x4a00001,
    parent 0x4a00001, window 0x4a00002, (10,10), width 50, height 50
border_width 4, override NO

PropertyNotify event, serial 14, synthetic NO, window 0x4a00001,
    atom 0x12b (WM_PROTOCOLS), time 326861, state PropertyNewValue

MapNotify event, serial 15, synthetic NO, window 0x4a00001,
    event 0x4a00001, window 0x4a00002, override NO

ConfigureNotify event, serial 20, synthetic NO, window 0x4a00001,
    event 0x4a00001, window 0x4a00001, (0,0), width 178, height 178,
    border_width 0, above 0xc00014, override NO

PropertyNotify event, serial 20, synthetic NO, window 0x4a00001,
    atom 0x1ec (_NET_WM_ALLOWED_ACTIONS), time 326861, state PropertyNewValue

ReparentNotify event, serial 20, synthetic NO, window 0x4a00001,
    event 0x4a00001, window 0x4a00001, parent 0xc0025a,
    (0,0), override NO

PropertyNotify event, serial 20, synthetic NO, window 0x4a00001,
    atom 0x1b2 (_NET_WM_DESKTOP), time 326863, state PropertyNewValue

PropertyNotify event, serial 20, synthetic NO, window 0x4a00001,
    atom 0x1b2 (_NET_WM_DESKTOP), time 326863, state PropertyNewValue

PropertyNotify event, serial 20, synthetic NO, window 0x4a00001,
    atom 0x174 (_NET_FRAME_EXTENTS), time 326863, state PropertyNewValue

ConfigureNotify event, serial 20, synthetic NO, window 0x4a00001,
    event 0x4a00001, window 0x4a00001, (11,36), width 178, height 178,
    border_width 0, above 0x0, override NO

PropertyNotify event, serial 20, synthetic NO, window 0x4a00001,
    atom 0x134 (WM_STATE), time 326864, state PropertyNewValue

PropertyNotify event, serial 20, synthetic NO, window 0x4a00001,
    atom 0x157 (_NET_WM_STATE), time 326864, state PropertyNewValue

ConfigureNotify event, serial 33, synthetic YES, window 0x4a00001,
    event 0x4a00001, window 0x4a00001, (54,66), width 178, height 178,
    border_width 2, above 0x0, override NO

MapNotify event, serial 33, synthetic NO, window 0x4a00001,
    event 0x4a00001, window 0x4a00001, override NO

VisibilityNotify event, serial 33, synthetic NO, window 0x4a00001,
    state VisibilityUnobscured

Expose event, serial 33, synthetic NO, window 0x4a00001,
    (0,0), width 178, height 10, count 3

Expose event, serial 33, synthetic NO, window 0x4a00001,
    (0,10), width 10, height 58, count 2

Expose event, serial 33, synthetic NO, window 0x4a00001,
    (68,10), width 110, height 58, count 1
[... 206 lines cut ...]
    root 0x14a, subw 0x0, time 333643, (911,591), root:(967,659),
    state 0x5, keycode 121 (keysym 0x1008ff12, XF86AudioMute), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

MappingNotify event, serial 48, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

MappingNotify event, serial 50, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyRelease event, serial 50, synthetic NO, window 0x4a00001,
    root 0x14a, subw 0x0, time 333690, (911,591), root:(967,659),
    state 0x5, keycode 105 (keysym 0xffe4, Control_R), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 51, synthetic NO, window 0x4a00001,
    root 0x14a, subw 0x0, time 333718, (911,591), root:(967,659),
    state 0x1, keycode 62 (keysym 0xffe2, Shift_R), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

ClientMessage event, serial 51, synthetic YES, window 0x4a00001,
    message_type 0x12b (WM_PROTOCOLS), format 32, message 0x12c (WM_DELETE_WINDOW)

kajotac, 2019-02-06:
I see the same exact information on my other Linux systems with the Kira. I tested on Debian, Arch and Ubuntu. I see the media commands (XF86AudioMute, XF86AudioPause, XF86AudioRaiseVolume and XF86AudioLowerVolume), but they are not getting catch by the system. Also, are the LED lights suppose to blink when Shift_R and Control_R are pressed?

schnikes, 2019-02-07:
> Also, are the LED lights suppose to blink when Shift_R and Control_R are pressed?
I'm getting the same. I assume it's meant to be like that.

haata, 2019-02-08:
Yes, the LEDs are supposed to blink/fade. It's shows which keys are on the layer. It can be disabled by setting the layer highlight fade (RAlt+RCtrl+F6).

kajotac, 2019-02-08:
Thank you for clarifying. Going back to my notes, when the Kira arrived and I plugged in media key combos were working fine, but I never notice the blink/fade effect to identify the layer. I only notice the effect with the 0.5.3 firmware. It's a nice idea and I'll keep it.

kkoltzau, 2019-02-12:
This change appears to break media keys for me on Windows 10. They work great when I revert eea5a4890fd537cc9b5651c28200b2b0dd8d64f5

haata, 2019-02-19:
This should be fixed in the upcoming v0.5.5 firmware. Did a lot of work on media keys.

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta

Download the K-Type bin file and use QuickFlash to load the firmware.
https://kiibohd.github.io/wiki/#/Configurator/Quickflash

kajotac, 2019-03-17:
I flash it! Does not work for me. Tested on a Ubuntu-based laptop. I even
started a video and I see the following in the mpv log: [input] No key
binding found for 'Shift+Ctrl+Volume_Up' or 'Shift+Ctrl+Volume_Down'. The
operating system doesn't do anything either when I press any of the media
combo keys. Will try in my arch system in a few minutes.

- Kennie Cruz

haata, 2019-03-18:
I'll need to load up Ubuntu on a system to check. 

Have you used any keyboards where media keys have worked in Linux? (in the past I've always had to manually set these up, even for USB remotes; but then again I haven't used stock Gnome or KDE in probably 10 years).

kajotac, 2019-03-18:
It also failed on my arch and Debian system. I tried my WhiteFox on Ubuntu,
arch and Debian, media combinations for volume up and down worked fine
(Fn+PgDn and Fn+PgUp). Is there a way I can revert back to original
firmware on the Kira? Just for testing purposes.

kajotac, 2019-03-18:
I will try my work Mac tomorrow with the Kira. Just in case is a problem
with th board.

haata, 2019-03-18:
You can either flash a layout with the configurator (not using QuickFlash), or you can flash the default firmware using the previous release using QuickFlash (https://github.com/kiibohd/controller/releases/tag/v0.5.4)

When was the last time you flashed your WhiteFox? (don't flash it, I have a suspicion that there may be a bug on Linux only that may have had a different key combination on WhiteFox).

kajotac, 2019-03-18:
The last time I flashed my WF was last year, but just for adding new
combinations. I think it has the same version since I got it. I’ll double
check.

kajotac, 2019-03-18:
That sounds reasonable. I tested my Kira on my work Mac and the media keys
work fine, same as the WF. Tested WF on Arch, Debian, Ubuntu, Fedora and
Mac. Media keys work. Did the same test with the Kira, only the Mac works.
What else can I do to help?

haata, 2019-03-18:
Thanks, on the WF for one of the Linux distros, can you do an xev trace while doing one of the media keys that does not work for Kira?
My current suspicion is actually due to the RCtrl and/or RShift also being held to get into the layer and that is confusing whatever is handling the media keys.
I may have to pull in a fix that I was planning for 0.6.0 to fix this particular issue.

kajotac, 2019-03-18:
Okay. I will.

kajotac, 2019-03-19:
I attached the xev output from the same system with Ubuntu, one for the
Kira and one for the WF. Hope it helps.

- Kennie Cruz

# cruzkj @ mandragora in ~ [8:12:42] 
$ xev
Outer window is 0x6400001, inner window is 0x6400002

PropertyNotify event, serial 8, synthetic NO, window 0x6400001,
    atom 0x27 (WM_NAME), time 7674204, state PropertyNewValue

PropertyNotify event, serial 9, synthetic NO, window 0x6400001,
    atom 0x22 (WM_COMMAND), time 7674204, state PropertyNewValue

PropertyNotify event, serial 10, synthetic NO, window 0x6400001,
    atom 0x28 (WM_NORMAL_HINTS), time 7674204, state PropertyNewValue

CreateNotify event, serial 11, synthetic NO, window 0x6400001,
    parent 0x6400001, window 0x6400002, (10,10), width 50, height 50
border_width 4, override NO

PropertyNotify event, serial 14, synthetic NO, window 0x6400001,
    atom 0x12e (WM_PROTOCOLS), time 7674205, state PropertyNewValue

MapNotify event, serial 15, synthetic NO, window 0x6400001,
    event 0x6400001, window 0x6400002, override NO

ConfigureNotify event, serial 20, synthetic NO, window 0x6400001,
    event 0x6400001, window 0x6400001, (0,0), width 178, height 178,
    border_width 0, above 0x2800015, override NO

PropertyNotify event, serial 28, synthetic NO, window 0x6400001,
    atom 0x1db (_NET_WM_ALLOWED_ACTIONS), time 7674205, state PropertyNewValue

ReparentNotify event, serial 29, synthetic NO, window 0x6400001,
    event 0x6400001, window 0x6400001, parent 0x280025e,
    (0,0), override NO

PropertyNotify event, serial 29, synthetic NO, window 0x6400001,
    atom 0x177 (_NET_FRAME_EXTENTS), time 7674209, state PropertyNewValue

ConfigureNotify event, serial 29, synthetic NO, window 0x6400001,
    event 0x6400001, window 0x6400001, (20,88), width 178, height 178,
    border_width 0, above 0x0, override NO

PropertyNotify event, serial 30, synthetic NO, window 0x6400001,
    atom 0x1b4 (_NET_WM_DESKTOP), time 7674284, state PropertyNewValue

PropertyNotify event, serial 31, synthetic NO, window 0x6400001,
    atom 0x137 (WM_STATE), time 7674285, state PropertyNewValue

PropertyNotify event, serial 31, synthetic NO, window 0x6400001,
    atom 0x15a (_NET_WM_STATE), time 7674285, state PropertyNewValue

ConfigureNotify event, serial 33, synthetic YES, window 0x6400001,
    event 0x6400001, window 0x6400001, (38,87), width 178, height 178,
    border_width 2, above 0x0, override NO

PropertyNotify event, serial 33, synthetic NO, window 0x6400001,
    atom 0x15a (_NET_WM_STATE), time 7674316, state PropertyNewValue

MapNotify event, serial 33, synthetic NO, window 0x6400001,
    event 0x6400001, window 0x6400001, override NO

VisibilityNotify event, serial 33, synthetic NO, window 0x6400001,
    state VisibilityUnobscured

Expose event, serial 33, synthetic NO, window 0x6400001,
    (0,0), width 178, height 10, count 3
[... 894 lines cut ...]
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 34, synthetic NO, window 0x6400001,
    mode NotifyGrab, detail NotifyAncestor

FocusIn event, serial 34, synthetic NO, window 0x6400001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 34, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 34, synthetic NO, window 0x6400001,
    mode NotifyGrab, detail NotifyAncestor

FocusIn event, serial 34, synthetic NO, window 0x6400001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 34, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

ClientMessage event, serial 34, synthetic YES, window 0x6400001,
    message_type 0x12e (WM_PROTOCOLS), format 32, message 0x12f (WM_DELETE_WINDOW)

kajotac, 2019-03-21:
I will go back to 0.5.4, I think there is a lightning issue with the beta
0.5.5. I will wait until its out and verify if its a real issue.
- Kennie Cruz

haata, 2019-03-22:
My current theory with this is that Linux doesn't like Ctrl and Shift being held along with the media keys.

I am working on a feature to automatically exclude layer activation keys (isolated expressions) for v0.6.0. Isolation is nearly ready but it's not tested enough to pull in for v0.5.5.
Moving this to v0.5.x

Also try binding one of the media keys in the configurator to the default layer (so you don't have to press ctrl+shift). My guess is that this will work.

kajotac, 2019-03-23:
I remember there was a kernel parameter for some especial key combinations,
I will check on a box by recompiling the kernel with and without the
option.

etrigan63, 2019-04-12:
> Also try binding one of the media keys in the configurator to the default layer (so you don't have to press ctrl+shift). My guess is that this will work.
I remapped the media keys to the base layer and they work now under Fedora 29. However, using Shift as a modifer for the Layer is causing the issue. I remapped Home to layer 1 and while my cursor indeed moved to home, it selected all of the text. I am going to to remap the layer shift combo so it does not use the Shift keys and will report back.

etrigan63, 2019-04-12:
I remapped the Layer 1 shift macro to the LGUI key and the media keys work as expected. The only niggle left is that tapping LGUI no longer brings up the start menu. Is there a way to map the LGUI so tap means LGUI and hold means Layer 1 Shift?

haata, 2020-03-19:
This firmware may help: https://github.com/kiibohd/controller/releases/tag/v0.5.7-beta2

kajotac, 2020-03-20:
Worked for me! Thank you.

- Kennie Cruz

haata, 2020-03-24:
Configurator should be updated now, just need to reflash your Kira from the configurator.

closed 2020-03-24 by haata