kiibohd/controller #325: Kira Media Keys Dont Work Unless Window is Focused

KaimTime, 2019-02-08:
* **Device**: Kira Metal Frame (Kickstarter) just received it yesterday

* **Firmware Origin**: Kiibohd Configurator Desktop Application

* **OS**: Windows 10 Pro 64 bit Build 16299.371

* **Version**: kira.v0.5.4

* **Reproduction Steps**: Running iTunes & attempting to control media with the media keys (play/pause, skip, previous track etc) all do not work unless iTunes is the focused application

* **Resulting Issue**: Controlling media w/ the media keys does not function unless the application is in focus. Note that windows volume & muting still works via media keys at all times.

haata, 2019-02-13:
Interesting.

I'm not actually sure how media keys are supposed to behave on Windows. I know long in the past for Winamp (does that even exist anymore?!) I had to set special global hotkeys, otherwise they'd only work in the active window.

KaimTime, 2019-02-13:
I know that iTunes complies with global media keys now but used to need the same setup.
It's definitely something in the way this keyboard handles media keys

haata, 2019-02-13:
Are there any keyboards I can reference where media keys do work?

KaimTime, 2019-02-13:
My MK DISCO I was using before this worked (and still does, tested it to make sure it wasnt a windows error and not the kira)

KaimTime, 2019-03-10:
any update on this being fixed? It's extremely annoying that you're unable to use the media keys as intended.

haata, 2019-03-10:
I have a fix that may possibly fix this for the next firmware release. I'm hoping to have a release out sometime on Tuesday/Wednesday.

KaimTime, 2019-03-11:
Great news, thanks for replying!

haata, 2019-03-13:
I'm going to be release a v0.5.5 beta out in the next day or so (trying to squeeze a few more fixes in).
Going to close this issue for now (I believe it's fixed), but feel free to re-open if the new firmware doesn't help.

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta

Download the K-Type bin file and use QuickFlash to load the firmware.
https://kiibohd.github.io/wiki/#/Configurator/Quickflash

KaimTime, 2019-03-18:
Just downloaded it & flashed to test it:
Same problem as before; Windows Volume & Muting works but the all the media player buttons (play/pause, next/prev track etc) do not work unless the media player is the window in focus still.

haata, 2019-03-22:
My current theory with this is that Linux doesn't like Ctrl and Shift being held along with the media keys.

I am working on a feature to automatically exclude layer activation keys (isolated expressions) for v0.6.0. Isolation is nearly ready but it's not tested enough to pull in for v0.5.5.
Moving this to v0.5.x

Also try binding one of the media keys in the configurator to the default layer (so you don't have to press ctrl+shift). My guess is that this will work.

KaimTime, 2019-03-22:
Yep binding them to not need to use the shortcut layer seems to have worked for Windows 10.
Hopefully the future fix you have will allow the shortcut keys + media keys to work!

KaimTime, 2019-05-09:
Any updates on a fix for this being implimented by the next patch?

KaimTime, 2020-03-14:
?

haata, 2020-03-14:
I'm thinking through a way to do this. I may have an idea that I'll try this weekend.

haata, 2020-03-14:
I have something working! It may not work for everyone, but should work well enough to enable by default.

Basically, I've added a layer trigger that blocks RCtrl and RShift when the function layer is active. Those keys will activate for a brief moment so applications that may use the combination as a trigger for something (VMWare Player?) should still function correctly.
(I've also added a define so that the functionality can be disabled in the configurator).

I'll try to push a beta release today that you can try out.

haata, 2020-03-19:
Please try this firmware to see if it helps: https://github.com/kiibohd/controller/releases/tag/v0.5.7-beta2

KaimTime, 2020-03-19:
looks like it's working!!! thank you so much

haata, 2020-03-19:
I'll leave this open until I push the new build to the configurator (should be in the next day or so).

haata, 2020-03-24:
Configurator firmware update has been pushed.

closed 2020-03-24 by haata