kiibohd/controller #326: Light color is not saved between plugging and unplugging

qubbit, 2019-02-09:
Light setting is not saved between plugging and unplugging.

* **Device**: Kira (Kono Store November 2018)

* **Firmware Origin**: GitHub release

* **OS**: macOS Mohave, Windows 10

* **Version**: 0.54

* **Reproduction Steps**: Change light color (R Ctrl + R Alt + F1) and save the setting (R Ctrl + R Alt + F12), unplug keyboard and plug it back.

* **Resulting Issue**: Light setting is not saved

haata, 2019-02-13:
I thought I fixed this. This may be a configurator issue, but I'll confirm that.

ericqzeng, 2019-02-13:
Can confirm that the v0.5.4 firmware for Kira works fine with the saving/loading/default functions. It seems that applying a custom config via the configurator is causing the issue.

gokieks, 2019-02-15:
I have this issue, though it doesn't manifest in the exact same way.  I can save the setting via RCtrl+RShift+F12, but when I either unplug/replug the keyboard or load it using RCtrl+RShift+F11, I get all the LEDs flashing instead of either what I saved or the original default.  A video of it can be seen here: https://youtu.be/AHH0I9o3O2w

Running 0.5.4 with stock layout.

gokieks, 2019-02-15:
So I decided to re-flash it again (still with default layout), and now it's somewhat closer to, but still not quite, working - I can save the settings and it will load (both via RCtrl+RShift+F11 and unplug/replug), except the color is not the one I saved, but rather the previous one in the cycle (when using RCtrl+RAlt+F1/F2) for both top and bottom lighting.

For every color except the last one in the cycle (Kira Blue, coincidentally the one I prefer to use) I can just save it with the next color, but for that one when I try saving it with the next color (which should be Blood Orange but doesn't actually change from Kira Blue and then just goes to Peach), saving and loading gets me a weird RGB setting instead.

haata, 2019-03-13:
This has been fixed for v0.5.5, you'll need to make sure for the color rotation animations (keys and underlighting) that `replace:clear` is changed to `replace:clearactive` in the configurator.

closed 2019-03-13 by haata