kiibohd/controller #327: WhiteFox TrueFox - LED Options Do Nothing

xnaas, 2019-02-09:
* **Device**: WhiteFox TrueFox (Massdrop 2016)

* **Firmware Origin**: Not sure exactly what it's asking for here, tbh. I'll go with web+desktop configurator

* **OS**: Windows 10 (1709, 1803)

* **Version**: ??

* **Reproduction Steps**:

1. Select a Key from a Layer
2. Choose REASSIGN
3. Select any LED control (LED+, LED-, LED OFF, LED ON)
4. Flash the firmware

* **Resulting Issue**: Absolutely no affect on LEDs.

-----

Unfortunately, whatever fix was implemented in #227 seems to not have helped?

Here is a zip that has a config from 2016 that *can* be flashed and has working LED controls: https://cdn.discordapp.com/attachments/325093040500768779/543933364302905346/WhiteFox-TrueFoxBlank-9c56b811454f14e81a4cbc6c170d51d8.zip

If you try to compile that after importing it, though: https://cdn.discordapp.com/attachments/325093040500768779/543934173505912874/failed-compile.txt

It's apparent that my keyboard requires 3 arguments to control the LEDs, but the configurator only supports 2 arguments.

xnaas, 2019-02-10:
Well, after tinkering around for another little while, I can't say what really changed, but I was able to get LED TOG, LED+, LED-, and LED OFF to work. + and - don't allow you to hold them like my 2016 config does, so going to max or minimum brightness definitely takes awhile (many key presses!) but I guess it 80% works?

haata, 2019-02-18:
Yes, the number of arguments changed (the LED control has changed..a lot, and the old arguments no longer made sense).

I had a lot of complaints about the repeat rate when adjusting brightness, so I removed that.
It should be easy to add that back once I add support for state scheduling.
https://github.com/kiibohd/controller/milestone/5

xnaas, 2019-02-18:
> I had a lot of complaints about the repeat rate when adjusting brightness, so I removed that.

Fair enough!

closed 2019-02-18 by xnaas