kiibohd/controller #329: Kira: can only disable backlighting via brightness?

ohaimark, 2019-02-11:
* **Device**:
Kira (Kickstarter)

* **Firmware Origin**:
Configurator

* **OS**:
Windows 10; irrelevant.

* **Version**:
Current.

* **Reproduction Steps**:
Attempt to disable backlighting by...

1) Setting all LEDs to blank in a custom static profile
2) Disabling all startup animations

* **Resulting Issue**:
Rapid strobing found in broken lighting profiles.

Suggested that startup brightness selection is exposed via the GUI in lighting area.

romedtino, 2019-02-12:
Have you Tried RCTRL+RALT+DEL ? On the default profile, this is should call LED OFF.

I was also able to create a single color animation profile to RGB (0,0,0) which seems to have all the LEDs off.

Unless I'm not understanding your question...

haata, 2019-02-24:
Added per-fade profile brightness control
ae7bf0b711dfba5e77451cd2f269032f929a8657

`U":A" : fade_control(0, 2, 0);`
That expression will set key LED brightness to 0.

`KLL_LED_FadeBrightness[0] = 0;` will set key brightness to 0 by default.

Will be available in v0.5.5

closed 2019-02-24 by haata