kiibohd/controller #330: Cannot detect Kira in flash mode

davidwisniewski, 2019-02-12:
* **Device**:
Kira (out of the box)

* **Firmware Origin**:
as installed upon delivery

* **OS**:
Win10

* **Version**:
current as of 12/02/2019

* **Reproduction Steps**:
Unboxed the Kira and followed the setup guide on the website (https://kiibohd.github.io/wiki/#/Keyboards/Kira): Unplug your keyboard, Install the Windows Drivers (Necessary to flash properly.), Plug back in your keyboard, Run the Firmware Updater, Enter flash mode by pressing Right Shift + Right Control + Esc, or by pressing the reset button on the bottom of the board (tried both)
The orange flash mode led is on. 

* **Resulting Issue**:
kira-firmware-v0.5.3-installer gives error message: "no DFU device found". 
The same issue arises when using the configurator. Downloading firmware gives error message "No keyboard in found in flash mode" even though the flash LED is on.

haata, 2019-02-13:
You may want to re-installing the drivers https://github.com/kiibohd/kiidrv/releases/download/v1.5.3-kiidrv/KiibohdDrivers.msi
For whatever reason sometimes it takes two tries.

davidwisniewski, 2019-02-13:
Thanks for the quick suggestion. I tried to both repair the drivers as well as remove+reinstall them. Had no effect though. Still cannot detect the Kira in flash mode.

davidwisniewski, 2019-02-13:
I am sorry for the close/open. That was actually my Kira going haywire...

Nothatso, 2019-02-13:
I'm having the same issue. I even tried 6 different USB ports on my PC and continue to get the same error when trying to get the firmware updater to recognize the Kira in flash mode.

I also noticed that when I plug in my KBD75 keyboard after trying to run the Kira firmware updater that it no longer functions properly unless I restart my PC.

Nothatso, 2019-02-13:
So I uninstalled then reinstalled the drivers, this time with my anti-virus disabled. The previous attempt at installing showed no indication that it failed. However, after disabling my AV, the program went through several additional steps in the cmd window before finishing. I then ran the firmware updater and it successfully identified the Kira in flash mode and updated. My PC then began acting like some of the Kira keys were being held down, but this was fixed by flashing the standard layout in the configurator. So far it appears to work. I hope this helps.

davidwisniewski, 2019-02-14:
Thank you, this worked perfectly. I uninstalled my AV, then uninstalled and reinstalled the Kira drivers. Indeed, the installation went through some additional steps that I did not see before. Now, the configurator detects the Kira in flash mode.

closed 2019-02-14 by davidwisniewski

haata, 2019-02-14:
This is very interesting. Thanks a lot @Nothatso (and @davidwisniewski for confirming!).
Would either of you mind sharing which AV software you have (so we can test the driver and flashing with AV in the future).

davidwisniewski, 2019-02-14:
I use Avira (current free version). I tried to switch off the AV by disabling "real-time protection", but that did not work. For me, only uninstalling the AV did the trick.

Nothatso, 2019-02-14:
I'm also using Avira, though disabling real-time protection worked for me.