kiibohd/controller #332: Wrong color for only one LED

kevinpoitra, 2019-02-13:
* **Device**:
Kira

* **Firmware Origin**:
GitHub release

* **OS**:
Windows 10 1809

* **Version**:
0.5.4

* **Reproduction Steps**:
N/A

* **Resulting Issue**:
One LED (the one under the 'x' key) seems to show an incorrect color instead of showing a color that matches the set animation. For example, using kira_wave, when the nearby keys are showing a pink color, the x key is blue. [Here's a video that shows the issue.](https://www.youtube.com/watch?v=vEjQD8RUqPs) 
I noticed this shortly after receiving the keyboard in early February - all I've done so far is compile the firmware myself (for 0.5.3, to help test the LED fault issue). I was gone for a week, and upon returning, I manually updated the firmware via the 0.5.4 release using the Configurator (without compiling it myself), and the issue persists.
Something to note - it's not just showing blue (as it appears in the video); if I change the animation to rainbow_wave, the x key's color does change, but not in line with the rest of the animation, [as seen in this second video.](https://www.youtube.com/watch?v=qM_bFnXssv0) It might be hard to tell, but it does also show a green color alongside blue, but that's it. It seems like that one key's animation is ahead of the rest of the keys, and that it is unable to show pink or red colors. The issue is even easier to see when setting a static color of pink across all keys, where the x key displays a blue color, [like in this image.](https://i.imgur.com/MWKCqNZ.jpg)

haata, 2019-02-13:
Thanks a ton for the videos and pictures! (they help a lot).

Can you try the color test animation: RCtrl+RAlt+Key Pad Enter

Most likely there is one color (red) that's not working.
If that's the case, the red channel on that LED is either broken or the solder joint may have lifted (there have been a few reports of this). If you're confident in your soldering abilities, it's an easy fix (just add a bit of solder to each of the pins on that LED). Otherwise you can contact support@kono.store and they'll help you out.

kevinpoitra, 2019-02-13:
No problem! I tried that test animation (which I didn't know about - seems to be very helpful). Green and blue show correctly for the x key, red looks like that LED is turned off, and white shows a cyan color. It definitely seems like a hardware defect rather than a firmware bug at this point (which I sort of assumed it would be considering it only affects one key across any animation, but you never know). I feel comfortable enough to perform the fix myself... but as luck would have it, my soldering iron is broken. I've placed an order for a new one and will report back here once I try adding some extra solder. Just to confirm, I'm not voiding my warranty by doing this, right?

haata, 2019-02-13:
Yeah, if you do it carefully it won't void your warranty.

It's possible that the Red channel is broken on the LED. I don't recommend trying to swap the LED as that is very tricky to do with destroying things (I don't have a good success rate and I solder a lot of surface mount things). In which case you can just contact support and they'll get you sorted out (link this github issue, it'll speed things along).

stoikos, 2019-02-23:
I have the exact same issue but under the 'b' key - just as in the above example, that LED will only show blue or green. I'm to know about the other reports: Is it always the Red channel that is affected?

mayorblurps, 2019-02-28:
@stoikos the LED/controller under my 3 key (row not numpad) is bad. Sometimes it works when an animation is going but 98% of the time I have my backlight set to a static color like white or light blue--3 LED sticks on magenta despite unplug/replug, but when the backlight is cyan or green the LED works fine at times. I emailed Kono support and sent mine back for warranty fix/replacement as I'm new to keebs and definitely would not feel comfortable trying to fix w solder.

stoikos, 2019-03-02:
@mayorblurps - Thanks for the info. A replacement would be too complicated for me at the moment. But I'd appreciate it if you could let me know how your replacement turns out, since there are a variety of hardware/assembly issues from what I've gathered.