kiibohd/controller #333: Kira - Play/Pause not working in macOS

jareddlc, 2019-02-14:
* **Device**:
Kira

* **Firmware Origin**:
Kiibohd Configurator 1.0.0

* **OS**:
macOS 10.14 (macbook pro 2015)

* **Reproduction Steps**:
a) Put laptop to sleep
b) Disconnect USB hub (Kira keyboard)
c) Go home, sleep, return to work :)
d) Wake laptop, connect keyboard

* **Resulting Issue**:
Certain media keys don't work, Play/Pause, Next/Prev. However, mute and volume do work. 

* **Workaround Issue**:
I've noticed that restarting my laptop while having the USB hub connected, will fix this issue.

I don't think this is a USB hub issue as I also own a another mech keyboard which does not experience this issue.

haata, 2019-02-18:
This should hopefully be resolved now in the upcoming v0.5.5

closed 2019-02-18 by haata

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta

Download the K-Type bin file and use QuickFlash to load the firmware.
https://kiibohd.github.io/wiki/#/Configurator/Quickflash

jareddlc, 2019-03-18:
@haata did you mean Kira bin?

jareddlc, 2019-03-18:
I think its working correctly, I've ran into some issues, but I think it is because of the spotify client.