kiibohd/controller #337: Static visualizations only work the first time they are compiled

planetxpress, 2019-02-18:
* **Device**:
Kira (Kono Store - January 2019)

* **Firmware Origin**:
kiibohd.github.io/wiki/#/Keyboards/Kira

* **OS**:
Windows 10 Pro 1803

* **Version**:
v0.5.4

* **Reproduction Steps**:

1. Create static LED visualization 
2. Configure your new animation as the Startup Animation
3. Flash keyboard
4. Close configurator
5. Open configurator and make an unrelated change, such as editing a keybinding on a layer
6. Flash keyboard
7. Keyboard backlighting turns red, instead of whatever static visualization you had created. You cannot get your static visualization to display again.

* **Resulting Issue**:
Static visualizations only work the first time that configurator compiles them. On any following compilation, the configurator seems to be mangling the JSON.

The affected line of the correct JSON with my static visualization working.
```
           "frames": [
                "### AUTO GENERATED - DO NOT EDIT - STATIC COLOR MAP ###",
                "P[1](0,0,0),\nP[2](0,0,0),\nP[3](0,0,0),\nP[4](0,0,0),\nP[5](0,0,0),\nP[6](0,0,0),\nP[7](0,0,0),\nP[8](0,0,0),\nP[9](0,0,0),\nP[10](0,0,0),\nP[11](0,0,0),\nP[12](0,0,0),\nP[13](0,0,0),\nP[14](0,0,0),\nP[15](0,0,0),\nP[16](0,0,0),\nP[17](0,0,0),\nP[18](255,255,255),\nP[19](0,0,0),\nP[20](0,0,0),\nP[21](255,255,255),\nP[22](0,0,0),\nP[23](0,0,0),\nP[24](255,255,255),\nP[25](0,0,0),\nP[26](0,0,0),\nP[27](255,255,255),\nP[28](0,0,0),\nP[29](0,0,0),\nP[30](255,255,255),\nP[31](0,0,0),\nP[32](0,0,0),\nP[33](255,255,255),\nP[34](0,0,0),\nP[35](0,0,0),\nP[36](0,0,0),\nP[37](0,0,0),\nP[38](0,0,0),\nP[39](0,0,0),\nP[40](0,0,0),\nP[41](0,0,0),\nP[42](0,0,0),\nP[43](0,0,0),\nP[44](0,0,0),\nP[45](0,0,0),\nP[46](0,0,0),\nP[47](0,0,0),\nP[48](0,0,0),\nP[49](255,255,255),\nP[50](0,0,0),\nP[51](0,0,0),\nP[52](0,0,0),\nP[53](0,0,0),\nP[54](0,0,0),\nP[55](0,0,0),\nP[56](0,0,0),\nP[57](0,0,0),\nP[58](0,0,0),\nP[59](0,0,0),\nP[60](0,0,0),\nP[61](0,0,0),\nP[62](0,0,0),\nP[63](0,0,0),\nP[64](0,0,0),\nP[65](255,255,255),\nP[66](0,0,0),\nP[67](0,0,0),\nP[68](0,0,0),\nP[69](0,0,0),\nP[70](0,0,0),\nP[71](0,0,0),\nP[72](0,0,0),\nP[73](0,0,0),\nP[74](0,0,0),\nP[75](0,0,0),\nP[76](0,0,0),\nP[77](0,0,0),\nP[78](0,0,0),\nP[79](0,0,0),\nP[80](0,0,0),\nP[81](0,0,0),\nP[82](255,255,255),\nP[83](0,0,0),\nP[84](0,0,0),\nP[85](255,255,255),\nP[86](255,255,255),\nP[87](0,0,0),\nP[88](255,255,255),\nP[89](255,255,255),\nP[90](255,255,255),\nP[91](0,0,0),\nP[92](255,255,255),\nP[93](0,0,0),\nP[94](0,0,0),\nP[95](255,255,255),\nP[96](0,0,0),\nP[97](0,0,0),\nP[98](255,255,255),\nP[99](0,0,0),\nP[100](255,255,255),\nP[101](0,0,0),\nP[102](0,0,0),\nP[103](0,0,0),\nP[104](255,255,255),\nP[105](0,0,0),\nP[106](255,255,255),\nP[107](0,0,0),\nP[108](255,255,255),\nP[109](0,0,0),\nP[110](0,0,0),\nP[111](0,0,0),\nP[112](0,0,0),\nP[113](255,255,255),\nP[114](0,0,0),\nP[115](255,255,255),\nP[116](0,0,0),\nP[117](0,0,0),\nP[118](255,255,255),\nP[119](0,0,0),\nP[120](255,255,255),\nP[121](0,0,0),\nP[122](255,255,255),\nP[123](0,0,0),\nP[124](0,0,0),\nP[125](0,0,0)"
            ],
```
The affected lines after restarting configurator with the static visualization no longer working.
```
           "frames": [
                "### AUTO GENERATED - DO NOT EDIT - STATIC COLOR MAP ###\nP[1](0,0,0),\nP[2](0,0,0),\nP[3](0,0,0),\nP[4](0,0,0),\nP[5](0,0,0),\nP[6](0,0,0),\nP[7](0,0,0),\nP[8](0,0,0),\nP[9](0,0,0),\nP[10](0,0,0),\nP[11](0,0,0),\nP[12](0,0,0),\nP[13](0,0,0),\nP[14](0,0,0),\nP[15](0,0,0),\nP[16](0,0,0),\nP[17](0,0,0),\nP[18](255,255,255),\nP[19](0,0,0),\nP[20](0,0,0),\nP[21](255,255,255),\nP[22](0,0,0),\nP[23](0,0,0),\nP[24](255,255,255),\nP[25](0,0,0),\nP[26](0,0,0),\nP[27](255,255,255),\nP[28](0,0,0),\nP[29](0,0,0),\nP[30](255,255,255),\nP[31](0,0,0),\nP[32](0,0,0),\nP[33](255,255,255),\nP[34](0,0,0),\nP[35](0,0,0),\nP[36](0,0,0),\nP[37](0,0,0),\nP[38](0,0,0),\nP[39](0,0,0),\nP[40](0,0,0),\nP[41](0,0,0),\nP[42](0,0,0),\nP[43](0,0,0),\nP[44](0,0,0),\nP[45](0,0,0),\nP[46](0,0,0),\nP[47](0,0,0),\nP[48](0,0,0),\nP[49](255,255,255),\nP[50](0,0,0),\nP[51](0,0,0),\nP[52](0,0,0),\nP[53](0,0,0),\nP[54](0,0,0),\nP[55](0,0,0),\nP[56](0,0,0),\nP[57](0,0,0),\nP[58](0,0,0),\nP[59](0,0,0),\nP[60](0,0,0),\nP[61](0,0,0),\nP[62](0,0,0),\nP[63](0,0,0),\nP[64](0,0,0),\nP[65](255,255,255),\nP[66](0,0,0),\nP[67](0,0,0),\nP[68](0,0,0),\nP[69](0,0,0),\nP[70](0,0,0),\nP[71](0,0,0),\nP[72](0,0,0),\nP[73](0,0,0),\nP[74](0,0,0),\nP[75](0,0,0),\nP[76](0,0,0),\nP[77](0,0,0),\nP[78](0,0,0),\nP[79](0,0,0),\nP[80](0,0,0),\nP[81](0,0,0),\nP[82](255,255,255),\nP[83](0,0,0),\nP[84](0,0,0),\nP[85](255,255,255),\nP[86](255,255,255),\nP[87](0,0,0),\nP[88](255,255,255),\nP[89](255,255,255),\nP[90](255,255,255),\nP[91](0,0,0),\nP[92](255,255,255),\nP[93](0,0,0),\nP[94](0,0,0),\nP[95](255,255,255),\nP[96](0,0,0),\nP[97](0,0,0),\nP[98](255,255,255),\nP[99](0,0,0),\nP[100](255,255,255),\nP[101](0,0,0),\nP[102](0,0,0),\nP[103](0,0,0),\nP[104](255,255,255),\nP[105](0,0,0),\nP[106](255,255,255),\nP[107](0,0,0),\nP[108](255,255,255),\nP[109](0,0,0),\nP[110](0,0,0),\nP[111](0,0,0),\nP[112](0,0,0),\nP[113](255,255,255),\nP[114](0,0,0),\nP[115](255,255,255),\nP[116](0,0,0),\nP[117](0,0,0),\nP[118](255,255,255),\nP[119](0,0,0),\nP[120](255,255,255),\nP[121](0,0,0),\nP[122](255,255,255),\nP[123](0,0,0),\nP[124](0,0,0),\nP[125](0,0,0)"
            ],
```

planetxpress, 2019-02-18:
Workaround I have found is to ironically edit out the line that says do not edit, then it works as expected.

monkeysontoast, 2019-03-24:
I am having the same exact issue. On Mac, just downloaded the configurator today. Any time I flash the keyboard the static color I created reverts to a reddish-orange.