kiibohd/controller #338: LEDs flashing wildly on startup

benkwa, 2019-02-20:
Device: Kira (Kono Store Sept 2018 -- delivered Jan 2019)

* **Firmware Origin**:
Generated by configurator 1.0.0 appimage

* **OS**:
Ubuntu 16.04.5
OSX 10.13.6 (High Sierra)

* **Version**:
bcdDevice            4.5f

* **Reproduction Steps**:
Unplug the keyboard, then plug it back in.  Alternatively, with the keyboard plugged in, close and re-open the laptop.
Repro'd on a Dell XPS 13 running Ubuntu 16.04.5.
Also repro'd on a Macbook Pro running OSX 10.13.6 (High Sierra)

* **Resulting Issue**:
Keyboard LEDs (both the key and underglow) flash wildly and cycle rapidly through all colors.
The key flashing stops if I push Ctrl+Alt+F1 and the underglow flashing stops if I push Ctrl-Alt-F2, then the LEDs return to a steady color.

benkwa, 2019-02-20:
[image]
Not quite capturing the effect due to frame-rate limiting on the GIF, but this captures the general idea.
Toward the end you see the LEDs fade out and then in, that's the "breathing" effect which apparently still works.

haata, 2019-03-10:
Sorry it's taken so long to respond, flu season for my family :| (all better now though!)

This is a bug partially with configurator, and partially in the firmware. I'm hoping it'll all be fixed up for the v0.5.5 firmware.
If you have a static color animation (cycles through colors) selected as an autostart animation, it'll cause what you see in your video. I'm still working on a full fix for this (because sometimes you do want to start a static color animation).
The other way if you don't select any animations for autostart (with the intention of having a blank keyboard). I've fixed this bug and it'll be released with v0.5.5

haata, 2019-03-13:
I've fixed this, took quite a bit of work, but should be working for v0.5.5.

closed 2019-03-13 by haata

benkwa, 2019-03-22:
I can confirm this is fixed.  Thank you!