kiibohd/controller #346: USB port disabled on K-Type after customization of firmware

timesscar, 2019-02-28:
I'm having an issue where if I create a new firmware with more color animations, and reflash my board, the right USB port stops working.  If I put the board into reflash mode the port starts working for that purpose again.  If I reflash the board back to the stock firmware, everything works normally too.

* **Device**:
 * K-Type (Massdrop November 2017)

* **Firmware Origin**:
Kiibohd configurator

* **OS**:
Windows 10 Version 1809

* **Version**:

* **Reproduction Steps**:
Use the configurator to create any animation (in this case I used the Miami wave built in pattern).  reflash the device.  The USB port is disabled.  Reflash the device with the default firmware (The one marked Standard in the UI) and the port is restored.

* **Resulting Issue**:
USB port is disabled for normal use (works for reflashing)

timesscar, 2019-02-28:
Here is the json as requested

[kiibohd-json.txt](https://github.com/kiibohd/controller/files/2916281/kiibohd-json.txt)

timesscar, 2019-02-28:
An update, I went through a testing cycle again, and I was incorrect above, the stock firmware from the kiibohd configurator does not fix the problem.  However, the one from the online configurator does.

haata, 2019-03-13:
Confirmed not working, still investigating a fix for this.

haata, 2019-03-15:
Fixed, will be in v0.5.5

closed 2019-03-15 by haata

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta

Download the K-Type bin file and use QuickFlash to load the firmware.
https://kiibohd.github.io/wiki/#/Configurator/Quickflash