kiibohd/controller #347: Media Keys don't work on Ubuntu

joshku, 2019-03-06:
* **Device**:
Kira

* **Firmware Origin**:
Kiibohd Configurator 1.0.0

* **OS**:
Ubuntu 18.04.1 LTS

* **Hardware**:
MacBook Pro (Retina, 15-inch, Mid 2015)

* **Version**:
0.4.2

* **Reproduction Steps**:
1. Plug in Kira keyboard laptop
1. Flash keyboard
1. Press `Right Shift` + `Right Control` + `PgDn`

* **Resulting Issue**:
Nothing happens. I tried it with `Home`, `End` and `PgUp` and nothing happens.

I have actually tried it on another linux machine running the same OS and I can't get the media keys to work as well.

haata, 2019-03-06:
I'm hoping to have this fixed with the next firmware release.
I've done a ton of testing on various OSs this time around.

Are you trying to use the volume keys primarily? You may want to check with `xev` to see if that's getting through.

joshku, 2019-03-06:
I just used `xev` and tried the control and got this 

```
KeyPress event, serial 45, synthetic NO, window 0x3200001,
    root 0x6db, subw 0x0, time 29580295, (1479,1007), root:(3428,1300),
    state 0x10, keycode 105 (keysym 0xffe4, Control_R), same_screen YES,
    XLookupString gives 0 bytes:
    XmbLookupString gives 0 bytes:
    XFilterEvent returns: False

KeyPress event, serial 45, synthetic NO, window 0x3200001,
    root 0x6db, subw 0x0, time 29581101, (1479,1007), root:(3428,1300),
    state 0x14, keycode 62 (keysym 0xffe2, Shift_R), same_screen YES,
    XLookupString gives 0 bytes:
    XmbLookupString gives 0 bytes:
    XFilterEvent returns: False

KeyPress event, serial 45, synthetic NO, window 0x3200001,
    root 0x6db, subw 0x0, time 29582022, (1479,1007), root:(3428,1300),
    state 0x15, keycode 121 (keysym 0x1008ff12, XF86AudioMute), same_screen YES,
    XLookupString gives 0 bytes:
    XmbLookupString gives 0 bytes:
    XFilterEvent returns: False
```

I think based on this it does recognize that I pressed the mute key but its not reacting to it.

haata, 2019-03-13:
I'm going to close this for now, I will be making a beta release of v0.5.5 within the next day or so. Hopefully the USB fixes I've been doing will help with the hotkeys.

It's been a long time since I used a standard desktop environment, but in the past most media keys didn't work out of the box. So I worry that might be the case here as well.
Do you have another keyboard where the media keys do work?

joshku, 2019-03-13:
@haata I do have two other keyboards that had working media keys. Primarily the media keys I used was the audio keys and they were able to increase/decrease/mute volume out of the box with nothing extra that needed to be installed.

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta

Download the K-Type bin file and use QuickFlash to load the firmware.
https://kiibohd.github.io/wiki/#/Configurator/Quickflash

joshku, 2019-03-18:
I didn't work for me :cry:

Although `xev` is still registering the events

```
KeyRelease event, serial 50, synthetic NO, window 0x6200001,
    root 0x6db, subw 0x0, time 4691351, (1147,737), root:(3096,1030),
    state 0x15, keycode 123 (keysym 0x1008ff13, XF86AudioRaiseVolume), same_screen YES,
    XLookupString gives 0 bytes:
    XFilterEvent returns: False
```

but nothing changed on my system. 

By the way, I am using the Kira keyboard and not the K-Type keyboard. Apparently flashing a k-type firmware onto a kira keyboard doesn't work :joy:

haata, 2019-03-22:
My current theory with this is that Linux doesn't like Ctrl and Shift being held along with the media keys.

I am working on a feature to automatically exclude layer activation keys (isolated expressions) for v0.6.0. Isolation is nearly ready but it's not tested enough to pull in for v0.5.5.
Moving this to v0.5.x

Also try binding one of the media keys in the configurator to the default layer (so you don't have to press ctrl+shift). My guess is that this will work.

simplexion, 2019-03-26:
What is happening is that Linux sees the media key being pressed but includes the Ctrl+Alt. If you change the media keymapping to Ctrl+Alt+Mute it will work. This is a workaround and not a fix.

joshku, 2019-04-08:
So I just remapped and added `ctrl` + `shift` before pressing `audio mute` and it works with the keyboard now. I am going to remap the rest but looks like this workaround works!

joshku, 2019-04-08:
Although I don't think it can differentiate between a right `ctrl` and a left `ctrl` but I guess it will do for now

tobiebooth, 2020-02-05:
@haata any update on when a fix for this will be pulled into a release?

haata, 2020-03-19:
Please try this firmware to see if it helps: https://github.com/kiibohd/controller/releases/tag/v0.5.7-beta2

haata, 2020-03-24:
Should be working now.

closed 2020-03-24 by haata