kiibohd/controller #348: Cluster of LED's Not working with custom Fingerprint Animation

rynodigital, 2019-03-08:
* **Device**: Kira

* **Firmware Origin**:Kiibohd configurator

* **OS**:macOS Mojave

* **Version**:10.14.3

* **Reproduction Steps**:
I wanted to have an animation, where by default all of the LED's are off, but when you press a key it lights up, and then fades to black.  I've created a **single_color** animation where the LED's are all black (#000000) and then i've got a secondary **fingerprint_two_tone** animation where the second color is also black, so that it fades out after it's pressed.

* **Resulting Issue**:
For whatever reason, the there is a cluster of four keys that don't respond to this.  The down arrow, right arrow, numpad 0, and numpad . are not working with this setup.  The LED's DO however work with any other animation, so i know they're not defective.

Link to video of it in action:
https://www.dropbox.com/s/9wg341x648ssgy5/IMG_0529.mov?dl=0

haata, 2019-03-10:
@rynodigital can you post the JSON file for your configuration? I think there may be a bug in the configurator generating the code but I'll need to check. Those are the last keys on the keyboard (Keypad Enter is on the previous row), so it's likely just some sort of limit error (should be easy to fix).

rynodigital, 2019-03-11:
I've got the JSON file uploaded here.  Thanks for the help on this!

[kira_config.txt](https://github.com/kiibohd/controller/files/2953140/kira_config.txt)

haata, 2019-03-13:
Aha, that's the problem:

` "custom-kll": "S[0x00-0x5F] :+ A[${__NAME__}](start);"`

Should be from 0x01-0x63 (1-99).
I'm going to find out if this actually needs to be fixed from inside the configurator.

rynodigital, 2019-03-13:
That seemed to fix it!  I was able to edit it within the Custom KLL section of the Configurator

Thanks for helping on this!

smasher816, 2019-03-13:
Updated for the configuration. Cheers!

https://github.com/smasher816/kiisrv/commit/432908a890e16771dbe0404224ac994748a35344

closed 2019-03-13 by haata