kiibohd/controller #349: Ergodox Infinity Only one hand works at a time

cogliostro, 2019-03-12:
Device: Ergodox
Firmware Origin:Kiibohd configurator
OS: Fedora
Version: 29

Reproduction Steps:
Flash each hand with either firmware from configurator or compiled from controller repo.

Resulting Issue:
Only one the device directly connected to the computer works. Both work if i connect them independently to the computer.

haata, 2019-03-13:
I'll take a look at this. I thought this was working.

This was working before? You haven't changed the interconnect cable?

cogliostro, 2019-03-13:
Same cable as before. Just new firmware.

dfaught, 2019-03-13:
I've been experiencing this for a while.  The last working commit I've been able to use for the Ergodox was:  https://github.com/kiibohd/controller/commit/449d4d9cd5097f129b0ee9b565cab656f8ae282f

cogliostro, 2019-03-13:
Can confirm that commit works for me too.

haata, 2019-03-16:
Confirmed broken. Also confirmed its an issue with the master (not the slave devices).

haata, 2019-03-17:
Should be fixed now. Make sure to flash both sides.

closed 2019-03-17 by haata

haata, 2019-03-17:
If you'd like to verify, you can try the beta: https://github.com/kiibohd/controller/releases/tag/v0.5.5-beta